yaourdt/mgos-to-tasmota #2: Shelly1 Initial Test - Not working

digiblur, 2020-06-25:
First of all, great idea and great job on this so far!  

I had a spare Shelly1 around so I flashed it back to the stock Shelly1 firmware and tried the procedure once I had it on my network.  I had a ping session open on it to see if it was going to reboot and such but it seems right after I send the URL it stops pinging for a couple seconds then comes back.  So I waited for the 4-5 minutes and refreshed the Shelly1 page, same stock firmware.  I tried it again after updating to the latest Shelly1 firmware but I got the same result.  Is there some sort of log I can check to see what's going on?  View it via serial maybe?

yaourdt, 2020-06-25:
Thank you for reporting this! I did not yet have a chance to test the OTA url and upload of the firmware, and this looks like the update failing during this stage.

You can establish a serial connection using [this guide](https://tasmota.github.io/docs/devices/Shelly-1/#serial-connection).

At this stage, the update either fails due to download issues or because the Shelly does not accept the firmware file. Can you try this url please?

http://shellyip/ota?url=https://github-production-release-asset-2e65be.s3.amazonaws.com/273783835/b93a8500-b6d9-11ea-8156-c30e89ec3ba6?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200625%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200625T133846Z&X-Amz-Expires=300&X-Amz-Signature=a7f7222a278b1218def8bdc9626832f4263267d5f3ea57ea8d7bc4745026e78a&X-Amz-SignedHeaders=host&actor_id=46676914&repo_id=273783835&response-content-disposition=attachment%3B%20filename%3Dfw.zip&response-content-type=application%2Foctet-stream

If this does not work either, console output would be very helpful

digiblur, 2020-06-25:
Looks like the same thing.  There doesn't seem to be much on the logs.  

state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : null

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 1880, room 16 
tail 8
chksum 0xf6
load 0x3ffe8000, len 824, room 0 
tail 8
chksum 0x99

digiblur, 2020-06-25:
Going to try it via local http server.

digiblur, 2020-06-25:
Putting the fw.zip file on a local HTTP nginx server seems to get further.  It spews a ton of stuff on the serial log (I assume the zip file contents going across).  But then after 30-45 seconds it comes back to the stock Shelly firmware.

yaourdt, 2020-06-25:
Sorry for my delayed response. Could you please send me the full logs for both attempts?

digiblur, 2020-06-25:
No rush...  it's a hobby for most.  ;)  I'll see if I can pull it on the second attempt as it was making Termite crash with the data

yaourdt, 2020-06-25:
@digiblur -thank you! in the meantime i managed to reproduce the problem.

@rojer - sorry for asking you again, but it seems to be that there is something fundamental i'm missing:

OTA updates from shelly stock version v1.3.0 to v1.7.0 are not a problem, and OTA flashing your shelly-homekit firmware also works as expected. 

Trying to OTA update this firmware, however, it seems that the shellys (or my dev-board with shelly stock firmware) either do not download the firmware (github does a 302 redirect before download to a very long url, see https://github.com/yaourdt/mgos-to-tasmota/issues/2#issuecomment-649547819 ), or it downloads it but rejects it for some reason.

> [Jun 25 20:35:11.808] pm open,type:0 0
> [Jun 25 20:35:58.350] state: 5 -> 0 (0)
> [Jun 25 20:35:58.350] rm 0
> [Jun 25 20:35:58.350] pm close 7
> [Jun 25 20:35:58.354] del if0
> [Jun 25 20:35:58.354] usl
> [Jun 25 20:35:58.354] mode : null
> [Jun 25 20:35:58.455] 
> [Jun 25 20:35:58.455]  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> [Jun 25 20:35:58.459] 
> [Jun 25 20:35:58.459] load 0x40100000, len 1880, room 16 
> [Jun 25 20:35:58.461] tail 8
> [Jun 25 20:35:58.464] chksum 0xf6
> [Jun 25 20:35:58.464] load 0x3ffe8000, len 824, room 0 
> [Jun 25 20:35:58.467] tail 8
> [Jun 25 20:35:58.467] chksum 0x99
> [Jun 25 20:35:58.471] csum 0x99 
>   +k      AP ZV  A Q Y  JR   u    - ) C
>     YNQ 5Yh  R           o  r  g|  d    d c     < s  d '   '   $`   s l $  [   : softAP(86:f3:eb:5a:
> [Jun 25 20:35:58.658] bcn 0
> [Jun 25 20:35:58.658] del if1
> [Jun 25 20:35:58.658] usl
> [Jun 25 20:35:58.658] mode : null
> [Jun 25 20:35:58.658] mode : sta(84:f3:eb:5a:17:ff)
> [Jun 25 20:35:58.660] add if0
> [Jun 25 20:35:58.663] sleep disable
> [Jun 25 20:35:58.663] WPA2 ENTERPRISE VERSION: [v2.0] disable
> [Jun 25 20:35:58.679] 
> [Jun 25 20:35:58.679] 
> [Jun 25 20:35:58.679] SHSW-1-5A17FF 84:F3:EB:5A:17:FF

thus, i tried updating via POST-request from shelly-homekit to this firmware. result:

> [Jun 25 20:14:45.556] shelly_main.cpp:258     Uptime: 43.64, RAM: 32856, 23720 free
> [Jun 25 20:14:45.660] mgos_http_server.c:180  0x3fff650c HTTP connection from 10.10.10.46:50550
> [Jun 25 20:14:45.660] mgos_http_server.c:188  0x3fff650c GET /favicon.ico
> [Jun 25 20:14:45.804] mgos_vfs.c:1026         Unmounting filesystems
> [Jun 25 20:14:45.807] state: 5 -> 0 (0)
> [Jun 25 20:14:45.810] rm 0
> [Jun 25 20:14:45.810] pm close 7
> [Jun 25 20:14:45.810] del if0
> [Jun 25 20:14:45.810] usl
> [Jun 25 20:14:45.810] mode : null
> [Jun 25 20:14:45.813] mgos_system.c:42        Restarting 
> [Jun 25 20:14:45.816]  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> [Jun 25 20:14:45.819] 
> [Jun 25 20:14:45.819] load 0x40100000, len 1880, room 16 
> [Jun 25 20:14:45.825] tail 8
> [Jun 25 20:14:45.825] chksum 0xf6
> [Jun 25 20:14:45.825] load 0x3ffe8000, len 824, room 0 
> [Jun 25 20:14:45.828] tail 8
> [Jun 25 20:14:45.831] chksum 0x99
> [Jun 25 20:14:45.831] csum 0x99 
>  T5j   5/J    [  PY Q  JZAEa
>    ! +a  ^ cL^X%o J )  IM %[ B  [   !m  JZA
> [Jun 25 20:14:45.842] \ ! Y !  I Q   
> [Jun 25 20:14:45.890]  ~      9      K T)  NY    

PRESS RESET BUTTON

> [Jun 25 20:14:53.820]    X 1  #RR   T *r  !          PA
> [Jun 25 20:14:53.823]   NY    "RL  T        9    @
[... 57 lines cut ...]
> [Jun 25 20:14:54.753]     - Service discovery
> [Jun 25 20:14:54.753] shelly_main.cpp:141     === Accessory not provisioned
> [Jun 25 20:14:54.759] mgos_init.c:36          Init done, RAM: 32856 total, 24808 free, 24816 min free
> [Jun 25 20:14:54.781] mgos_mongoose.c:66      New heap free LWM: 24768
> [Jun 25 20:14:54.785] mgos_ota_core.c:1318    UID: 03ee41545dbb4fc0, license: none
> [Jun 25 20:14:54.795] mgos_mongoose.c:66      New heap free LWM: 23888
> [Jun 25 20:14:54.798] mgos_mdns.c:84          Listening on udp://:5353
> [Jun 25 20:14:54.805] lwip_mdns.c:39          Joining multicast group 224.0.0.251
> [Jun 25 20:14:54.812] mgos_net.c:86           WiFi STA: connecting
> [Jun 25 20:14:55.761] shelly_main.cpp:258     Uptime: 1.61, RAM: 32856, 25072 free
> [Jun 25 20:14:56.760] shelly_main.cpp:258     Uptime: 2.61, RAM: 32856, 24872 free
> [Jun 25 20:14:57.503] scandone
> [Jun 25 20:14:58.386] state: 0 -> 2 (b0)
> [Jun 25 20:14:58.386] shelly_main.cpp:258     Uptime: 4.23, RAM: 32856, 24872 free
> [Jun 25 20:14:58.398] state: 2 -> 3 (0)
> [Jun 25 20:14:58.405] state: 3 -> 5 (10)
> [Jun 25 20:14:58.405] add 0
> [Jun 25 20:14:58.405] aid 3
> [Jun 25 20:14:58.405] cnt`

my build command is `mos build --build-var MODEL=Shelly1 --platform esp8266 --build-var BOARD=esp8266-2M` (i tried the command with and without `BOARD=esp8266-2M`).

using `mos flash` with my firmware works just fine. 

i'm not sure what exactly i'm missing at this point. i'll try OTA from a minimal firmware next, using ota-http-server

yaourdt, 2020-06-25:
a minimal OTA won't work either. 

> 
> [Jun 25 21:14:19.647] mgos_ota_http_serve:139 MG_EV_HTTP_PART_DATA: filedata fw.zip 379
> [Jun 25 21:14:19.653] mgos_ota_core.c:728     switch1.bin: CRC OK
> [Jun 25 21:14:19.659] mgos_ota_core.c:265     Update state 5 -> 6
> [Jun 25 21:14:19.661] mgos_ota_core.c:265     Update state 6 -> 4
> [Jun 25 21:14:19.667] mgos_ota_http_serve:139 MG_EV_HTTP_PART_DATA: filedata fw.zip 379
> [Jun 25 21:14:19.673] mgos_ota_http_serve:175 MG_EV_HTTP_PART_END: filedata fw.zip 0
> [Jun 25 21:14:19.679] mgos_ota_http_serve:191 MG_EV_HTTP_MULTIPART_REQUEST_END: 0
> [Jun 25 21:14:19.682] mgos_ota_core.c:397     CDFH for switch1-0.2.1/manifest.json
> [Jun 25 21:14:19.688] mgos_ota_core.c:397     CDFH for switch1-0.2.1/esp_init_data_default_v08.bin
> [Jun 25 21:14:19.696] mgos_ota_core.c:397     CDFH for switch1-0.2.1/fs.bin
> [Jun 25 21:14:19.701] mgos_ota_core.c:397     CDFH for switch1-0.2.1/rboot.bin
> [Jun 25 21:14:19.706] mgos_ota_core.c:397     CDFH for switch1-0.2.1/switch1.bin
> [Jun 25 21:14:19.710] mgos_ota_core.c:424     EOCD
> [Jun 25 21:14:19.713] mgos_ota_core.c:629     Reached the end of archive
> [Jun 25 21:14:19.719] mgos_ota_core.c:265     Update state 4 -> 7
> [Jun 25 21:14:19.751] esp_ota_backend.c:330   New rboot config: prev_rom: 0, current_rom: 1 current_rom addr: 0x200000, current_rom size: 605088, current_fs addr: 0x48000, current_fs size: 262144
> [Jun 25 21:14:19.765] mgos_ota_core.c:785     0: 1 1
> [Jun 25 21:14:19.768] mgos_ota_core.c:265     Update state 7 -> 8
> [Jun 25 21:14:19.771] mgos_ota_core.c:811     Update finished, result 1 (Update applied, rebooting)
> [Jun 25 21:14:19.780] esp_ota_backend.c:452   cur 1 prev 0 fwu 1
> [Jun 25 21:14:19.782] mgos_vfs.c:283          updater.dat -> /updater.dat pl 1 -> 1 0x3ffeff6c (refs 1)
> [Jun 25 21:14:19.796] mgos_vfs.c:377          open updater.dat 0x0 0x1b6 => 0x3ffeff6c updater.dat -1 => -1 (refs 0)
> [Jun 25 21:14:19.802] mgos_event.c:135        ev OTA1 triggered 1 handlers
> [Jun 25 21:14:19.808] mgos_ota_core.c:852     Update requires reboot
> [Jun 25 21:14:19.814] mgos_utils.c:38         Rebooting in 500 ms
> [Jun 25 21:14:20.319] mgos_event.c:135        ev MOS2 triggered 0 handlers
> [Jun 25 21:14:20.323] mgos_vfs.c:1026         Unmounting filesystems
> [Jun 25 21:14:20.327] mgos_vfs_dev.c:203      root refs 1
> [Jun 25 21:14:20.330] mgos_vfs_dev.c:203      root refs 0
> [Jun 25 21:14:20.333] mgos_vfs_dev.c:203      sfl0 refs 1
> [Jun 25 21:14:20.336] mgos_vfs_dev.c:203      sfl0 refs 0
> [Jun 25 21:14:20.339] state: 5 -> 0 (0)
> [Jun 25 21:14:20.342] rm 0
> [Jun 25 21:14:20.342] pm close 7
> [Jun 25 21:14:20.342] del if0
> [Jun 25 21:14:20.342] usl
> [Jun 25 21:14:20.344] mode : null
> [Jun 25 21:14:20.344] mgos_system.c:42        Restarting 
> [Jun 25 21:14:20.347]  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> [Jun 25 21:14:20.353] 
> [Jun 25 21:14:20.353] load 0x40100000, len 1540, room 16 
> [Jun 25 21:14:20.356] tail 4
> [Jun 25 21:14:20.356] chksum 0x82
> [Jun 25 21:14:20.356] load 0x3ffe8000, len 748, room 4 
> [Jun 25 21:14:20.362] tail 8
> [Jun 25 21:14:20.362] chksum 0x65
> [Jun 25 21:14:20.362] csum 0x65
> [Jun 25 21:14:20.365] rBoot v1.2.1-cesanta1 - richardaburton@gmail.com
> [Jun 25 21:14:20.368] Flash Size:   32 Mbit
> [Jun 25 21:14:20.371] Flash Mode:   DIO
> [Jun 25 21:14:20.371] Flash Speed:  80 MHz
> [Jun 25 21:14:20.374] rBoot Option: Big flash
> [Jun 25 21:14:20.378] 
> [Jun 25 21:14:20.378] First boot, attempt 0
> [Jun 25 21:14:20.382] Boot is unconfirmed
> [Jun 25 21:14:20.431] Booting rom 1 (0x200000).
> [Jun 25 21:14:20.442] sp 0x3fffdd3c 
> [Jun 25 21:14:20.442] sf_dump[0] a0: 0x40002514  a1: 0x3fffdd3c  a2: 0x00000001  a3: 0x0028c010 
> [Jun 25 21:14:20.448] epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0xffffffef, depc=0x00000000
> [Jun 25 21:14:20.456] Fatal exception (28): 
> [Jun 25 21:14:20.459] epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x007d7ff0, depc=0x00000000
> [Jun 25 21:14:20.467] Fatal exception (28): 
> ...

digiblur, 2020-06-25:
Sorry, been a busy/crazy day with work and haven't able to circle back even during the lunch break.  I guess that would explain why the http pull via local nginx server worked a bit more.  

Is the download of the bin not working due to https or something with github?  Could we try to pull from this link?  http://thehackbox.org/tasmota/release/tasmota-lite.bin

rojer, 2020-06-25:
yes, HTTPS from github won't work because stock fw doesn't ship the necessary CA certificate.
that's why i'm using plain http to serve my firmware.

yaourdt, 2020-06-25:
Okay, i see. that explains the OTA via cloud problem. I does not explain the second issue. i just saw, that the stock shelly firmware uses another boot config address. trying this now.

rojer, 2020-06-25:
yes, boot config address needs to be changed to 0x2000, i forgot to include that in myPR

yaourdt, 2020-06-25:
yes, that seems to be it. but 0x1000, not 0x2000? thank you for all your help, now lets find a server that still serves http..

rojer, 2020-06-25:
yes, 0x1000, sorry.

yaourdt, 2020-06-25:
no worries. @digiblur , can you confirm that this build https://github.com/yaourdt/mgos-to-tasmota/releases/download/release-147866930/fw.zip works if you flash it from your local http server? it works on my dev board, but will it work on a true shelly?

digiblur, 2020-06-25:
I will try shortly. Storm just rolled through with high winds and knocked out the power.

yaourdt, 2020-06-25:
oh boy! i hope you are all safe!

digiblur, 2020-06-25:
All good except for about 5 or 6 battery backups singing their songs. Heard reports of a possible small tornado south of my location.

digiblur, 2020-06-25:
Well well well... you've done it!  I have a couple other models to "test".  :)

yaourdt, 2020-06-26:
nice \o/

i've also updated the OTA link, so this issue should be resolved. 

@digiblur i'll add the other shelly models over the next days, so you can test them if you want.

closed 2020-06-26 by yaourdt