yaourdt/mgos-to-tasmota #3: Supporting other Shelly devices

RavenSystem, 2020-07-10:
Could you include these devices too, please? They are supported by Tasmota and Home Accessory Architect:
- Shelly RGBW2.
- Shelly Dimmer.
- Shelly EM.

Thanks!

yaourdt, 2020-07-11:
I'll have to look into it. Mongoose OS is open-source, but the Shelly firmware built on top of it isn't, so I can't promise anything - but I think we can get all information from the official firmware images (preferably the ZIP files, not a bin). As I don't own any of the devices, can you provide me with those images, so I can build a test version for you?

RavenSystem, 2020-07-11:
URLs of all original Shelly firmwares can be found here: https://api.shelly.cloud/files/firmware

And these are direct link of devices:
- Shelly RGBW2: http://api.shelly.cloud/firmware/SHRGBW2_build.zip
- Shelly Dimmer 1: http://api.shelly.cloud/firmware/SHDM-1_build.zip
- Shelly Dimmer 2: http://api.shelly.cloud/firmware/SHDM-2_build.zip
- Shelly EM: http://api.shelly.cloud/firmware/SHEM_build.zip

yaourdt, 2020-07-11:
Nice, I didn't know about the JSON listing them all. Thank you! 

Unfortunately I cannot provide you with a test version of this firmware at the moment, as I usually build against the development branch for avoiding a bug in Mongoose OSes current version, but the development branch seems currently to be broken. I'll try again later today or tomorrow and let you know.

yaourdt, 2020-07-11:
Okay, I just tried building against the current version of MGOS. Please try

`http://shellyip/ota?url=http://dl.dasker.eu/test/shellyDimmer1.zip`

`http://shellyip/ota?url=http://dl.dasker.eu/test/shellyDimmer2.zip`

`http://shellyip/ota?url=http://dl.dasker.eu/test/shellyRGBW2.zip`

`http://shellyip/ota?url=http://dl.dasker.eu/test/shellyEm.zip`

and if they work, I'll add them to the next release. The builds contain the bug mentioned above, which will result in the update to take a very long time (up to 30 minutes?), but it will not fail. This problem arises due to faulty HTTPS connection termination in Mongoose OS 2.17.0, but I think we can accept this for our quick test.

Warning: Although it is unlikely, the test firmware might brick the device, so prepare to recover via a wired connection

rojer, 2020-07-11:
@yaourdt btw, master build is fixed

yaourdt, 2020-07-11:
@rojer - thank you for letting me know! i've updated the firmware links in my comment above, it should now work as expected. as soon as @RavenSystem will confirm the firmware images are ok, i'll add them to the supported systems list.

@rojer: is there a mongoose os developers mailing list or similar that i can join for discussions and announcements? or just the gitter channel? the reason i'm asking is that for the tasmota -> mongoose os firmware i encountered a problem i'd like to discuss, but don't fell it belongs here or in the tasmota issue, as it is an entirely different topic.

rojer, 2020-07-11:
@yaourdt there's https://community.mongoose-os.com/ but i personally prefer gitter (https://gitter.im/cesanta/mongoose-os). i have it open as a pinned tab and check regularly, there are others too who can answer your questions. feel free to ask questions there.

justinmartin, 2020-07-24:
I'd love support for the Shelly Lightbulbs as well.  Really looking forward to weekend project now.

yaourdt, 2020-07-25:
@justinmartin I'll have to look into it, but this should not be a problem. I'll see that I add some more Shelly devices tomorrow or on Monday. 

Everyone: If you have tested the devices mentioned above, please give me feedback, so I can add them to the main release. I'd dislike adding them without someone confirming they work and I don't have the dimmers etc. for testing.

digiblur, 2020-07-25:
Looks like they are still working on the Dimmers in this thread:  https://github.com/arendst/Tasmota/issues/6914

@yaourdt I did walkthrough of your process here - https://youtu.be/_oRr8FZyyQ0

yaourdt, 2020-07-27:
@justinmartin (ore anyone else), please try 

`http://shellyip/ota?url=http://dl.dasker.eu/test/shellyBulb.zip`
`http://shellyip/ota?url=http://dl.dasker.eu/test/shellyVintage.zip`

If confirmed that they are working, I'll add them to the main branch

justinmartin, 2020-07-27:
Thanks.  I ordered one last week and I'm waiting for it to come in.

yaourdt, 2020-08-13:
Closing this due to positive feedback on most of the mentioned devices.

digiblur, 2020-08-14:
We might need one for the Shelly US plug soon too ;)  Think mine should be here soon.

yaourdt, 2020-08-17:
@digiblur - is the firmware different from the PlugS? Because that is already supported :) If different: [Which of those](https://api.shelly.cloud/files/firmware) is it?

digiblur, 2020-08-17:
I don't have it in my hands yet, still in the mail, might be today or tomorrow.  I'm thinking this one looking at the list.  SHPLG-U1

I did order two, one for science ;)

digiblur, 2020-08-17:
@yaourdt I don't have it in my hands yet, still in the mail, might be today or tomorrow. I'm thinking this one looking at the list. SHPLG-U1

I did order two, one for science ;)

yaourdt, 2020-08-17:
Oh, very good 😁
I'll add SHPLG-U1 then, as soon as I figure out the HAA issue, which seem to be a bit more complicated than initially thought

yaourdt, 2020-08-17:
@digiblur done :)

digiblur, 2020-08-17:
> @digiblur done :)

Many thanks @yaourdt I'm hoping that is the one, guessing based on the U and this is the 1st US plug for them.  How does it all tie together based on the name and firmware?

yaourdt, 2020-08-18:
Basically there are two things we need to adapt for each model:
1. Shellys check the device name before performing an update. If it does not match, the update will be rejected. This is a fail-safe so users don't accidentally flash the wrong firmware. The device name is not the brand name, but something like eg. `shelly-plug-s` for the Shelly Plug S or `bulb6w` for the Shelly Vintage Bulb. So sometimes it is clear, sometimes I have to guess which firmware is the matching one.
2. The boot configuration for different models resides at different addresses. If not respected, the device will get stuck during early boot as it is looking for the config in the wrong place. Boot config can be found in the manifest file of the respective firmware.

This is why we need a different firmware file for every model and cannot have a single universal firmware update.

digiblur, 2020-08-20:
@yaourdt the plug worked perfectly!  Thanks!  One thing I would like to see is to use the 8.4.0 tasmota-lite.bin file to start.  No rush but I would happy to test on a Shelly1 to ensure everything goes perfectly.  I assume we need lite based on the bin file size.

srnoth, 2020-09-18:
@yaourdt any chance we could see firmware for the Shelly Duo bulb? Looks like it's called "SHBDUO" under the stock firmware listing. Would be happy to help test.

michelnet, 2020-10-08:
any chance we could see firmware for the Shelly i3.  The devices shoud by compatible with tasmota. -> a Template exits. https://templates.blakadder.com/shelly_i3.html

http://api.shelly.cloud/firmware/SHIX3-1.zip
I have two of them i can test it.

yaourdt, 2020-10-22:
@michelnet if you like to, please try http://dl.dasker.eu/firmware/mg2tasmota-ShellyI3.zip and tell me if i got it right :)

michelnet, 2020-10-22:
@yaourdt Flashing with this firmware works. Greate. Also the template if you have just one Button configured. With 3 buttons all in the same topic. I will check that what is wrong.

yaourdt, 2020-10-22:
good to hear! If you get the template figured out, I'd be happy to update it

mariowitdoek, 2020-11-14:
@yaourdt there seems to be a new tasmota confirmation for the shelly dimmer 1 and 2: https://github.com/arendst/Tasmota/commit/c02770ce965b07064133708b713f3034b0d69c5c

yaourdt, 2020-11-16:
@mariowitdoek do you have a template, so I can update the README.md? I don't have the dimmers myself and there is no template in the commit or the blakadder repo

digiblur, 2020-11-16:
@yaourdt check this commit from the man himself...  

https://github.com/arendst/Tasmota/commit/b6b35ef86e7e715787a273a9ff0c8c58afb475da

yaourdt, 2020-11-16:
Ups, I indeed could have found that. Thanks, @digiblur ! I've just updated the readme

digiblur, 2020-11-16:
No worries, we can't see it all.  We all appreciate your support in this project!

mariowitdoek, 2020-11-16:
Hei men

You are both amazing!

@digiblur KUTGV(ideos)! I love them. To the point and clearly explained. :)

Mario
Aka SledgehammerRulez(that 80's show with David rasche, the real dirty
Harry :D)

mariowitdoek, 2020-11-21:
I tested the shelly dimmer following this (https://github.com/arendst/Tasmota/issues/6914#issuecomment-723185441) and I got a shelly dimmer with tasmota. I think the template is different that the one mentioned above. But I will hold off until this tasmota development version is released.

yaourdt, 2020-11-23:
@mariowitdoek you are right, some GPIO definitions are different. I don't have a dimmer, so you'll have to tell me which (or maybe both?) templates work. Did you test the other template?

mariowitdoek, 2020-11-24:
I will pass them back ones when have acces to the shelly (i think saturday)

mariowitdoek, 2020-12-02:
hei mark

this is the template that I used and Theo Arendst also mention that the
first was not OK
{"NAME":"Shelly Dimmer
1","GPIO":[0,3200,0,3232,5504,5536,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}

[image: image.png]

[image: image.png]

radionerd1, 2020-12-21:
Thanks, I have a flaky shellyplugu1 that keeps locking up on shelly firmware, looses WiFi and freezes. OTA firmware worked great.
One little hiccup was my browser (firefox) retains shelly's WEB cache waiting for the switch to return. I could have cleared cache, but instead just added /? to the switch URL. (My Plug ip: 192.168.1.219**/?**)  Then it loaded the Tasmota web ui.

yaourdt, 2020-12-22:
@mariowitdoek - thank you for figuring this out! I've updated the README accordingly ✅

closed 2020-12-27 by yaourdt

mariowitdoek, 2021-01-02:
hello Mark,

once again tx for your tool to update my shelly devices. I just flashed 4
shellyRGBW2, 5 ShellyDimmer1 and 2 Shelly1.

Maybe some experiences:

*ShellyRGBW2*
I had trouble updating them and read the whole (closed) thread. The OTA url
was only working when I upgraded to the latest shelly firmware.
*"The current Firmware version of your Shelly device is
20190822-083406/master@4148d2b7 New version 20201124-092125/v1.9.0@57ac4ad8
is available."*

before the upgrade I got response:
{
"status": "updating",
"has_update": false,
"new_version": "20201124-092125/v1.9.0@57ac4ad8",
"old_version": "20190822-083406/master@4148d2b7"
}
with the OTA,
after the update I got
{
"status": "updating",
"has_update": false,
"new_version": "20201124-092159/v1.9.0@57ac4ad8",
"old_version": "20201124-092159/v1.9.0@57ac4ad8",
"beta_version": "20201202-140859/v1.9.3-rc3@50c6ab57"
}
And I checked that the 'color' mode was selected in the shelly settings.
Maybe interesting to mention this shelly firmware upgrade in the table or
in a FAQ.

One shelly was f*cked up when testing for a solution. It was getting an IP
but no webserver.
The fix was with uploading tasmota.bin with tasmotizer.
So newbies and persons with 2 left hands like me, do not have to worry:
tasmotizer can save the day :).

*Shelly Dimmer 1*
I have noticed that the template on the official tasmota page is little
different than on your github page.
I know that they correspond to the initial communication but maybe the
tasmota team has optimized it.

your site
{"NAME":"Shelly Dimmer
1","GPIO":[0,3200,0,3232,5504,5536,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
tasmota:
{"NAME":"Shelly Dimmer
1","GPIO":[0,3200,0,3232,5568,5600,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}

oh yeah, there is additional stm firmware that needs to be uploaded as well
(cfr https://templates.blakadder.com/shelly_Dimmer.html).
tx to digiblurdiy for mentioning it a couple of weeks ago.

Oh yeah, Happy New year with a lot of thinkering with tasmota! And
hopefully tuya gets back on his decision for his firmware (I don't buy them
any more).
I have bought 2 Athom lights with tasmota preflashed and they work
splendidly. Instant reaction and nice integration with the tasmota
integration addon in home assistant.
Before there was a tuya lamp (present from my wife :p) and it was sloooooow
(why does my request to put the light on needs a trip around the world to
china ... and back??).

Mario
aka SledgeHammerRulez (Trust me I know what I'm doing, with David Rasche as
SledgeHammer, the real Dirty Harry ;D)

yaourdt, 2021-01-04:
Hey Mario,

happy new year to you too, and thanks for all the input!

 * Shelly RGBW2: I've added a hint about updates.
 * Can you describe what happened step by step with this one? I'm really curious where to find out what caused this and why tasmotizer saved it.
> One shelly was f*cked up when testing for a solution. It was getting an IP
> but no webserver.
> The fix was with uploading tasmota.bin with tasmotizer.
> So newbies and persons with 2 left hands like me, do not have to worry:
> tasmotizer can save the day :).

 * Shelly Dimmer 1: According to your [comment above](
 https://github.com/yaourdt/mgos-to-tasmota/issues/3#issuecomment-731601367), the version with `{"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5504,...` should be the right one, or did I get that wrong? Additional Firmware: I need to start a wiki for this project I think, as the README will get too full of all those details

 * I did not know you can get tasmota preflashed. Is it just for the US?

mariowitdoek, 2021-01-04:
hi Mark

for the dimmer 1 ...
on your site the template is
   {"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5504,5536
,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
but on the tasmota site the template is:
   {"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5568,5600
,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
I discovered it by writing it all down for future reference.

for the RGBW2 ...
i tried the usual way but I got 2 ping losses and then back to shelly page
i had these devices already for a while and I tried the flash it but with
an old downloaded version on a webserver on my local network. I had one on
my drive from my previous flashing.
(http://192.168.2.170/ota?url=http://192.168.2.52/mg2tasmota-ShellyRGBW2.zip).
Initially, I did not want to update the shelly firmware (to prevent it from
refusing tasmota update) because I was thinking "hei my previous flashing
went well with this firmware ...".
However, this method resulted in a shellyRGBW2 that gets an IP but no
webpage (it even did not listen on port 80). I assume there was a fault in
the file mg2tasmota-ShellyRGBW2.zip that was on my drive.
hei, no worries. Since I got an IP, then I assumed the device was OK.
Shelly developers made the pin layout accessible so I have put some dupont
cables in it, grounded GPIOpin (easy, next to GND :D),  hooked it on my
FTDI converter of my laptop and flashed it using latest Tasmotizer. I think
I erased it first before uploading the latest firmware.
Then I tried the update to the latest firmware for the other shellies an
all was well for the update procedure.

for the preflashed lamp ...
The devices with tasmota out of the box are listed on
https://templates.blakadder.com/preflashed.html.
I have bought 2 from aliexpress and all is well ....
https://nl.aliexpress.com/item/4001281509550.html?spm=a2g0s.9042311.0.0.3da24c4dqLTK09
They came with version 8 and their name in the footer (not Theo Arends) but
after flashing to version 9, it was the real deal :).

For a home renovation project I needed some (16) multi gang wall switches.
At that time, the model of Athom only had 2 relays, so I choose to buy
Sonoffs T1 3 Gang.
Now they have a 4 gang ... :S (
https://templates.blakadder.com/athom_SW11-TAS-4EU.html)
Let's call it karma ... but now I have fun converting sonoffs with my FTDI
adapter (since SonOTA is no longer possible).

Mario

PS you have to tell me. I live in Belgium, the dutch speaking side of it
(near Brugge/Bruges) but we learned French in school. Does your name
reference "yogurt" or is it coincidence. I don't want to offend you in any
way but each time I see the video of digiblurDIY with the reference to your
github alias "that he can no pronounce", my brain makes that liaison :)

yaourdt, 2021-01-05:
Ok, I've updated the dimmer (and some other Shellys) templates according to templates.blakadder.com, it will be corrected with the next release.

For the RGBW2: Thanks for the description. Using outdated versions of this tool can be dangerous, as things changed a lot in the beginning, so for now I recommend using the latest version (things should get a bit more stable however). I'm glad Tasmotizer saved the day!

Yes, the name is indeed a (by purpose) misspelled version of the French word yaourt 😁