yaourdt/mgos-to-tasmota #8: Flashing OTA other ESP devices...

rlajos, 2020-08-13:
First of all thank you for great work, I helped me a lot!

I just would like to know is there a way to flash (OTA) ESP devices (other than Shelly) with this method?
The mgos-to-tasmota method would be even better for devices needed soldered GPIO access only. I have flashed in the past some MagicHome devices and it was really cumbersome...
[https://tasmota.github.io/docs/devices/MagicHome-with-ESP8285/](url)

Is there a way to build a kind of "general" *.bin and than customize it with corresponding template?

Thank you again,

Lajos

yaourdt, 2020-08-13:
Hi Lajos,

I can add more Shelly devices for you without much effort, but I cannot provide a single firmware file that fits all Shellys, due to the way they process updates. Similarly to the Shelly devices, it is not much effort to support any other device that is shipped with Mongoose OS, and I'm open to requests 😊

Unfortunately, mg2x will not work for devices running other types of firmware besides Mongoose OS. As these different firmware types all use different update procedures, there is no such thing as a universal firmware binary that fits all esp8266 devices.

As for the MagicHome device you linked: I'm not certain what kind of firmware they ship with, but if it is a Tuya device, have a look at [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert/), that does the same thing as mg2x, but for Tuya based devices.

rlajos, 2020-08-15:
Understand, thanks anyway...

closed 2020-08-17 by yaourdt