yaourdt/mgos-to-tasmota #10: Update fullhaaboot to the last version.

seritos, 2020-09-02:
Hi!!

So, first of all, thanx so much for this SW that you have developed. It is helping so much to the community to flash device. Almost everyone is using it to flash these devices. 

Regarding the update, would it be possible please to use the last version of fullhaaboot that has been published? Previous contain some bugs that might cause problems during the installation. 

https://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin

Last, but not least, would it be somehow possible that the version updates directly form the official repository? 

Thanx much!

yaourdt, 2020-09-24:
Hi @seritos,

sorry for my late reply! I'll be releasing an updated version of the installer with Tasmota 8.5 and HAA 3.4.0 within the next hours.

Automatically installing the latest version is theoretically possible, however I'd rather not do that [as explained here](https://github.com/yaourdt/mgos-to-tasmota/issues/7#issuecomment-674974148). Other tools such as tuya convert also perform version pinning for the same reasons. 

For Tasmota and espurna, this is not a problem, as the updating process is reliable. For HAA, I'll try to update more often from now on, until the project reaches a state where updates between different minor versions are safe. I think this is the least ugly solution for this dilemma. If you have a better idea to make it bleeding edge and safe at the same time, please let me know!

RavenSystem, 2020-09-27:
HAA OTA v4.2.1 has been released.

yaourdt, 2020-09-28:
@RavenSystem - thanks for the information. I'll test it now

closed 2020-09-28 by yaourdt

seritos, 2020-09-28:
Thanx @yaourdt!!