yaourdt/mgos-to-tasmota #11: Shelly Plug US

jnoxon, 2020-09-13:
I successfully flashed a Shelly Plug US with 8.5.0. Thanks!

If I set it as a Shelly 1, the basic functionality works--I can turn it on and off.

What's a good starting point so I can start to figure out how to add support for the power monitoring? I don't mind doing custom Tasmota builds, I just have no idea where to start.

I have a defective Shelly Plug US that I don't mind cutting open if it helps.

Thanks for any pointers!

jnoxon, 2020-09-13:
For what it's worth, I stumbled across the Shelly Plug S template, and it doesn't work at all. I'd really hoped the hardware would be internally similar.

```{"NAME":"Shelly Plug S","GPIO":[56,255,158,255,255,134,0,0,131,17,132,21,255],"FLAG":2,"BASE":45}```

jnoxon, 2020-09-13:
DigiBlur provided this template which seems to mostly work:

``` {"NAME":"ShellyPlugUS","GPIO":[52,255,57,255,21,134,0,0,131,17,132,157,0],"FLAG":0,"BASE":45} ```

closed 2020-09-15 by jnoxon

yaourdt, 2020-09-24:
@jnoxon sorry for not replying to you sooner! I'm happy you / @DigiBlur figured out a template, I'll add it to the README.md file, so others can use it too.

yaourdt, 2020-09-24:
Thank you for sharing it!

digiblur, 2020-09-24:
https://youtu.be/_TSJB_IzxG0

https://www.digiblur.com/2020/08/shelly-plug-us-with-power-monitoring.html