yaourdt/mgos-to-tasmota #12: Shelly T&H

jnoxon, 2020-09-15:
Is it possible to convert Shelly TH? I run mine on USB power,so the lack of sleep in Tasmota is not a concern. Thank you!

yaourdt, 2020-09-24:
Yes, I can add support for the SHHT with the next release, no problem. But you will have to build a Tasmota template for it in order to get it fully working.

jnoxon, 2020-09-24:
After disassembling one of my TH units, I think it's far from straightforward to support it with Tasmota. There's a Cortex M0 inside, which makes me conclude that the sensor is hanging off the M0 and not the ESP. The M0 must be used to read the sensor and wake the ESP. Reversing this would take some time and almost certainly require custom Tasmota changes.

Given there's a flashing header inside the TH, it might be a bad idea to make it software flash-able until someone has the motivation and skill to figure all of it out. Since I can get better results from a D1 mini and a Si7021 for a lot less money, I can't justify working on it.

yaourdt, 2020-09-24:
That sounds like a lot of effort indeed.

I decided to still add it in case anyone wants to have a try, but with a warning message next to it. There is now also a (slightly experimental) way back via [tasmota-to-mgos](https://github.com/yaourdt/tasmota-to-mgos), so the device won’t be permanently useless if anyone decides to go forward.

closed 2020-09-24 by yaourdt