yaourdt/mgos-to-tasmota #13: Shelly Duo Bulb

srnoth, 2020-09-23:
@yaourdt any chance we could see firmware for the Shelly Duo bulb? Looks like it's called "SHBDUO" under the stock firmware listing. Would be happy to help test.

http://api.shelly.cloud//firmware/SHBDUO-1.zip

yaourdt, 2020-09-24:
Yes, we can certainly do that. I'll add it to the next release and will be looking forward to your feedback 😊

srnoth, 2020-09-27:
@yaourdt thank you so much. The new version for Shelly Duo worked perfectly.

Here is a working template:
{"NAME":"Shelly Duo","GPIO":[0,0,0,0,38,37,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}

yaourdt, 2020-09-28:
Awesome, thank you for your feedback and for providing the template! I've added the template to the projects README.md for the next update.

closed 2020-09-28 by yaourdt