yaourdt/mgos-to-tasmota #14: Tasmota to mgos?

ashmodai6, 2020-11-15:
In the readme it mentions that there is no known way yet to come back from tasmota to the original shelly fw thru OTA but I in this conversation https://github.com/yaourdt/mgos-to-tasmota/issues/9 it seems very well be possible ?
Thanks !

yaourdt, 2020-11-16:
Yes, have a look at https://github.com/yaourdt/tasmota-to-mgos

Apart from my own devices and @pandabear41, there is not much feedback yet, so I'd still consider it unstable software. But still I could mention this in the README.md, you are right

yaourdt, 2020-11-16:
If you want to give it a try, please let me know if you had any issues or if the process worked as intended.

ashmodai6, 2020-11-17:
So I just successfully reverted two shelly 1 back from tasmota to the original shelly firmware without any issue. I have to go thru the extra step of upgrading to tasmota minimal as a first step but all was flawless. Thanks a lot ! I refrained to do so until now since I did want to enjoy dismantling my shellys from the wall :)

ashmodai6, 2020-11-18:
😀 actually i did it because it was well documented in your readme !

closed 2020-12-22 by yaourdt