yaourdt/mgos-to-tasmota #15: Unable to migrate RGBW2

Yarikx, 2020-11-22:
RGBW2 doesn't work for me with the default update link.
After opening the link I got a response without `new_version`:

```
{"status":"updating","has_update":false,"new_version":"","old_version":"20200827-070157/v1.8.3@4a8bc427"}
```

Looks like it refuses to parse/read firmware archive.

yaourdt, 2020-11-23:
@Yarikx can you please post the exact link you used?

Yarikx, 2020-11-26:
I used the link in the Readme.md, with IP replaced it was
`http://192.168.107.20/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyRGBW2.zip`

yaourdt, 2020-11-27:
I just tried it with firmware 20201124-092159/v1.9.0@57ac4ad8 on a nodemcu development board, and can confirm the error. I'll look into ist

Lamaras, 2020-12-07:
I confirm the error too.
`{"status":"updating","has_update":false,"new_version":"","old_version":"20201124-092235/v1.9.0@57ac4ad8"}`
@yaourdt, are there any news?

Bascht74, 2020-12-20:
Same problem here, unable to update...

yaourdt, 2020-12-21:
I had a look at it again, and it is just weird why it won't work for this specific device.

Usually, the device decides based on the firmware name given in the updates `manifest.json` file if the update is accepted or rejected. There are some other parameters that need to be set for the update to be successful, they are however not validated beforehand (and even if they were, they seem to be set correctly).

So the only current explanation I have is, that there is an additional check for this device that I'm missing completely. As I don't have the source for the Shelly devices, and there is no useful logging, frankly I'm a bit out of ideas. I'll try some more things in a deep dive tomorrow, but it is unlikely to work without further information.

Yarikx, 2020-12-22:
@yaourdt I have one RGBW2 that is not yet installed, so I can try to capture some logs. Hope it will help to diagnose the problem.

yaourdt, 2020-12-22:
Ok, small steps forward: Using the current version of mgos-to-tasmota, the update will start, resulting in 

`{"status":"updating","has_update":false,"new_version":"20201124-092159/v1.9.0@57ac4ad8","old_version":"20201124-092159/v1.9.0@57ac4ad8","beta_version":"20201202-140859/v1.9.3-rc3@50c6ab57"}`

but will not contiue after that. @Yarikx - I only have a dev-board, would you mind trying an update with your Shelly? Please attach a serial console, and post the logs, as well as the devices JSON response.

I'm looking for the output starting with the boot process before the update attempt until the boot process after the (failed) update, which should be similar to this:

```
[Dec 22 18:26:17.138]  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[Dec 22 18:26:17.141] 
[Dec 22 18:26:17.141] load 0x40100000, len 1732, room 16 
[Dec 22 18:26:17.145] tail 4
[Dec 22 18:26:17.147] chksum 0x89
[Dec 22 18:26:17.147] load 0x3ffe8000, len 832, room 4 
[Dec 22 18:26:17.150] tail 12
[Dec 22 18:26:17.150] chksum 0x7c
[Dec 22 18:26:17.154] csum 0x7c
[Dec 22 18:26:17.154] rBoot v1.2.1-cesanta1-yaourdt - richardaburton@gmail.com
[Dec 22 18:26:17.159] Flash Size:   32 Mbit
[Dec 22 18:26:17.162] Flash Mode:   DIO
[Dec 22 18:26:17.162] Flash Speed:  80 MHz
[Dec 22 18:26:17.165] rBoot Option: Big flash
[Dec 22 18:26:17.168] rBoot Option: move esp init data
[Dec 22 18:26:17.171] 
[Dec 22 18:26:17.171] Moving SDK config sectors before booting.
[Dec 22 18:26:17.174] First boot, attempt 0
[Dec 22 18:26:17.177] Boot is unconfirmed
[Dec 22 18:26:17.250] Booting rom 0 (0x8000).
[Dec 22 18:26:17.271] ...    mode : softAP(86:f3:eb:5
[Dec 22 18:26:17.669] bcn 0
[Dec 22 18:26:17.669] del if1
[Dec 22 18:26:17.669] usl
[Dec 22 18:26:17.669] mode : null
[Dec 22 18:26:17.669] mode : sta(84:f3:eb:5a:17:ff)
[Dec 22 18:26:17.672] add if0
[Dec 22 18:26:17.675] sleep disable
[Dec 22 18:26:17.675] WPA2 ENTERPRISE VERSION: [v2.0] disable
[Dec 22 18:26:17.684] 
[Dec 22 18:26:17.684] 
[Dec 22 18:26:17.684] SHRGBW2-5A17FF 84:F3:EB:5A:17:FF
[Dec 22 18:26:17.686] 
[Dec 22 18:26:20.514] scandone
[Dec 22 18:26:21.403] state: 0 -> 2 (b0)
[Dec 22 18:26:21.411] state: 2 -> 3 (0)
[Dec 22 18:26:21.419] state: 3 -> 5 (10)
[Dec 22 18:26:21.420] add 0
[Dec 22 18:26:21.420] aid 9
[Dec 22 18:26:21.420] cnt 
[Dec 22 18:26:21.429] 
[Dec 22 18:26:21.429] connected with XXX, channel 6
[Dec 22 18:26:21.432] dhcp client start...
[Dec 22 18:26:21.448] ip:192.168.178.101,mask:255.255.255.0,gw:192.168.178.1
[Dec 22 18:26:31.422] pm open,type:0 0
[Dec 22 18:28:30.139] state: 5 -> 0 (0)
[Dec 22 18:28:30.139] rm 0
[Dec 22 18:28:30.139] pm close 7
[Dec 22 18:28:30.143] del if0
[Dec 22 18:28:30.143] usl
[Dec 22 18:28:30.143] mode : null
[Dec 22 18:28:30.244] 
[Dec 22 18:28:30.244]  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[Dec 22 18:28:30.246] 
[Dec 22 18:28:30.246] load 0x40100000, len 1732, room 16 
[Dec 22 18:28:30.250] tail 4
[Dec 22 18:28:30.253] chksum 0x89
[Dec 22 18:28:30.253] load 0x3ffe8000, len 832, room 4 
[Dec 22 18:28:30.256] tail 12
[Dec 22 18:28:30.256] chksum 0x7c
[Dec 22 18:28:30.259] csum 0x7c
[Dec 22 18:28:30.259] rBoot v1.2.1-cesanta1-yaourdt - richardaburton@gmail.com
[Dec 22 18:28:30.264] Flash Size:   32 Mbit
[Dec 22 18:28:30.267] Flash Mode:   DIO
[Dec 22 18:28:30.267] Flash Speed:  80 MHz
[Dec 22 18:28:30.270] rBoot Option: Big flash
[Dec 22 18:28:30.273] rBoot Option: move esp init data
[Dec 22 18:28:30.277] 
[Dec 22 18:28:30.277] Moving SDK config sectors before booting.
[Dec 22 18:28:30.298] Booting rom 0 (0x8000).
[Dec 22 18:28:30.319] ... mode : softAP(86:f3:eb:5
[Dec 22 18:28:30.464] bcn 0
[Dec 22 18:28:30.464] del if1
[Dec 22 18:28:30.464] usl
[Dec 22 18:28:30.464] mode : null
[Dec 22 18:28:30.464] mode : sta(84:f3:eb:5a:17:ff)
[Dec 22 18:28:30.467] add if0
[Dec 22 18:28:30.470] sleep disable
[Dec 22 18:28:30.470] WPA2 ENTERPRISE VERSION: [v2.0] disable
[Dec 22 18:28:30.479] 
[Dec 22 18:28:30.479] 
[Dec 22 18:28:30.479] SHRGBW2-5A17FF 84:F3:EB:5A:17:FF
[Dec 22 18:28:30.481] 
[Dec 22 18:28:33.308] scandone
[Dec 22 18:28:34.197] state: 0 -> 2 (b0)
[Dec 22 18:28:34.202] state: 2 -> 3 (0)
[Dec 22 18:28:34.206] state: 3 -> 5 (10)
[Dec 22 18:28:34.206] add 0
[Dec 22 18:28:34.206] aid 9
[Dec 22 18:28:34.206] cnt 
[Dec 22 18:28:34.216] 
[Dec 22 18:28:34.216] connected with XXX, channel 6
[Dec 22 18:28:34.219] dhcp client start...
[Dec 22 18:28:34.234] ip:192.168.178.101,mask:255.255.255.0,gw:192.168.178.1
[Dec 22 18:28:44.208] pm open,type:0 0
```

Yarikx, 2020-12-22:
Made an attempt to update rgbw2 with the serial attached:
```
pm open,type:0 0
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : null

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 1884, room 16
tail 12
chksum 0x4b
ho 0 tail 12 room 4
load 0x3ffe8000, len 824, room 12
tail 12
chksum 0x24
csum 0x24�ԩu��jUJ�B��TTTTj-�VVU���U�(TT�U��␋������EJ�KP�T�����Q��Z�U�JZ�E�EZQԩu�],�Vը����K��婵Z�V�UP��XQU�TV2
Mo
ph������␛�n��p��o|�␄l���sl␜b���␀␌�|␃r��$�N�␄�o�␀␄l ␃��r�$�$␒�␟␀␌␄�mode : softAP(b6:e6:2d:5a:39:6f)
add
bcn 0
del if1
usl
mode : null
mode : sta(b4:e6:2d:5a:39:6f)
add if0
sleep disable
WPA2 ENTERPRISE VERSION: [v2.0] disable

SHRGBW2-5A396F B4:E6:2D:5A:39:6F

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 11
cnt

connected with Little Wing, channel 1
dhcp client start...
ip:192.168.107.27,mask:255.255.255.0,gw:192.168.107.1
pm open,type:0 0
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : null

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 1884, room 16
tail 12
chksum 0x4b
ho 0 tail 12 room 4
load 0x3ffe8000, len 824, room 12
tail 12
chksum 0x24
csum 0x24�ԩu��jUJ�B��TTTTj-kVVU���U�(TT�U��␋�������TT�U+UZ�I�UEJ�KP␕�JZ�E��]Qԩu�],��ը�������ũu��V�UP��XQU�TV2
Mo
ph����␃�␒�N�{��o|�␌d���␎$␜c���␀␄��␒{��l�o�␌�o�␃␌$`␂��r�d�␜�␂�␄␌␄�mode : softAP(b6:e6:2d:5a:39:6f)
add
bcn 0
del if1
usl
mode : null
mode : sta(b4:e6:2d:5a:39:6f)
add if0
sleep disable
WPA2 ENTERPRISE VERSION: [v2.0] disable

SHRGBW2-5A396F B4:E6:2D:5A:39:6F

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 11
cnt

connected with Little Wing, channel 1
dhcp client start...
ip:192.168.107.27,mask:255.255.255.0,gw:192.168.107.1
```

Json output:
```json
{
  "status": "updating",
  "has_update": false,
  "new_version": "20201124-092159/v1.9.0@57ac4ad8",
  "old_version": "20201124-092159/v1.9.0@57ac4ad8",
  "beta_version": "20201202-140859/v1.9.3-rc3@50c6ab57"
}
```

yaourdt, 2020-12-23:
Thank you, @Yarikx ! I'll test something and get back to you

yaourdt, 2020-12-27:
Ok, I fixed the 1L, but the RGBW2 is still not working. 

@rojer - for some reason, the RGBW2 rejects this firmware update as invalid. Would you know a difference compared to the other Shelly devices that could explain this behavior? As I don't have the source for the Shelly devices, and there is no debug logging, I'd really appreciate a hint if you have the time

Logs of failed update attempts are in the comment above, the mos.yml contains this as device specific section:
```
  - when: build_vars.MODEL == "ShellyRGBW2"
    apply:
      name: rgbw2
      build_vars:
        FS_SIZE: 262144
        FLASH_SIZE: 2097152
        BOOT_CONFIG_ADDR: 0x7000
        MGOS_ROOT_FS_TYPE: SPIFFS
```
The latest test build is here: http://dl.dasker.eu/rbgw/firmware.zip

Bascht74, 2020-12-27:
Can you change the following in your generated zip file:
1. change the folder from `rgbw2-0.3.1` to `rgbw2-1.0` and at the end `"version": "0.3.1"` to `"version": "1.0"`
2. delete the cs_sha256 everywhere as they are not listed in the original firmware
3. Insert       "fs_block_size": 4096, "fs_erase_size": 4096 and "fs_page_size": 256, 

in the original firmware it looks like this: 
```
"fs": {
      "addr": 765952, 
      "addr1": 1814528, 
      "cs_sha1": "2ee0425e40a27f50010159c6ed159b81cb42d38d", 
      "fs_block_size": 4096, 
      "fs_erase_size": 4096, 
      "fs_page_size": 256, 
      "fs_size": 262144, 
      "size": 262144, 
      "src": "fs.bin", 
      "type": "fs"
```

I think it can only be those little things. If the SHA1 is OK, it can be only in the metadata and that is a little different. Maybe the RGBW2 has some extra checks in...

Sebastian

yaourdt, 2020-12-27:
@Bascht74 Good thinking, but changing the version unfortunately makes no difference. I've updated the file at ` http://dl.dasker.eu/rbgw/firmware.zip` accordingly, you are welcome to test it. Also, Shelly firmware versions and Mongoose OS versions are not matching. Unsolved question: How to set the Shelly version?

For you other suggestions: The manifests are auto-generated by the mos build tool, which also creates the zip archive. You are welcome to edit them and repack in the right order. It will most likely not make a difference, as the additional file system parameters are default values. The frozen lib (mongooses json parser), which I assume is used in the update mechanism as well, will ignore additional keys such as _cs_sha256_.

There are also some more parameters, which cannot be seen in the manifest.json file, as for example (here the Shelly 1L)
```
cdefs:
  PRODUCT_HW_REV: '"1.0"'
  STOCK_FW_MODEL: '"SHSW-L"'
  PRODUCT_VENDOR: '"Allterco"'
  PRODUCT_MODEL: '"Shelly1L"'
```
and a file called _hwinfo_struct.json_ in the devices file system which might be of relevance. 

Further research is needed....

Bascht74, 2020-12-27:
> and a file called _hwinfo_struct.json_ in the devices file system which might be of relevance.
OK, then it was to easy to guess. But to be shure, it might be a good idea to manually tweak the files so that at this stage there is not difference. 
The only two possibilies that I can think off are that the firmware update process is not started (the manifest files, version and so on) or that it is canceled in the first stage (maybe hwinfo_struct or something else). 
Does the log say anything about where which stage the update process is canceled?

Sebastian

rojer, 2020-12-28:
i have not worked on the RGBW2 yet, but one thing i know is that it has different firmware versions for color and white modes, and they have different app names: rgbw2-color and rgbw2-white respectively.

yaourdt, 2020-12-28:
@rojer that was the missing bit of information, thank you! The update worked on my nodemcu with firmware name set to rgbw2-color.

@Yarikx does this work on the actual device as well? `http://dl.dasker.eu/rbgw/firmware.zip`

Yarikx, 2020-12-28:
@yaourdt It actually worked!
Was able to upgrade rgbw2!
Thanks a lot!

yaourdt, 2020-12-28:
Nice! I'll add the fix to the next release

closed 2020-12-28 by yaourdt