yaourdt/mgos-to-tasmota #18: Shelly 1L

sanderlv, 2020-12-18:
Hi, first of all, wow! This is very cool!

I have successfully flashed multiple devices as a step toward esphome (sorry for that but for me more flexible with hass).

I wondered if the shelly 1L will als become in you list.

Its such a cool device "without neutral".

https://shelly.cloud/products/shelly-1l-single-wire-smart-home-automation-relay/

yaourdt, 2020-12-22:
You are welcome! I've added support for the Shelly 1L, if it works, can you please confirm this to me?

Update URL: `http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-Shelly1L.zip`

schrej-zz, 2020-12-23:
Hi,
I tested it and it seems to fail:
{"status":"updating","has_update":false,"new_version":"20201205-115619/v1.9.2_1L@8f181f76","old_version":"20201205-115619/v1.9.2_1L@8f181f76","beta_version":"20201202-135922/v1.9.3-rc3@50c6ab57"}

yaourdt, 2020-12-23:
Thank you @schrej-zz - it seems to be the same problem as for the RGBW2. I'll look into it

yaourdt, 2020-12-27:
@schrej-zz please try to update from `http://dl.dasker.eu/1l-test/firmware.zip`, which worked on the dev board

schrej-zz, 2020-12-28:
@yaourdt,
thanks. Works perfectly.

yaourdt, 2020-12-28:
Thank you for testing it! I've included it into the current release

closed 2020-12-28 by yaourdt

Bibinsa, 2021-02-08:
Hi !
Shelly 1L version 2.4 can't be converted to Tasmota this way.

AchilleTalon-Lefuneste, 2021-02-19:
Hi. I tasmotized my Shelly1L OTA, worked great. It works perfectly when using neutral (physical switch and web ui switch work as expected), but when using without neutral, I can't get wifi or switch to work: the light remains on at all times.
Any idea? Any similar experience ?