yaourdt/mgos-to-tasmota #19: Not working if static IP is set

jzeeman01, 2020-12-21:
Hi
Awesome OTA update!!
I did notice that in the shelly WIFI settings - if I set a static IP the update does not seem to work?
I have tested with a number of devices - the moment I remove the static IP - it updates perfectly

Rgds
J

yaourdt, 2020-12-21:
So basically, the update fails, and you are back to the Shelly original firmware?

In this case, please double-check your gateway and subnet mask settings when assigning the IP manually. Both the original Shelly firmware, and my intermediate, need internet access to download the next firmware. If you don't set a gateway together with the manual IP, the download will fail as the device cannot reach the respective download server.

jzeeman01, 2020-12-21:
That may be it as I left the default / optional settings - mask defaults to 255.255.255.0 and is fine but the default gateway (optional) was not populated
When I have another device to flash I'll populate it - might be some time though
Thx again for the awesome software
Rgds
J

yaourdt, 2020-12-21:
You are very welcome! I'll close this for now, as I think this was the problem. If it turns out not to be the solution, feel free to re-open and comment 👍

closed 2020-12-21 by yaourdt