yaourdt/mgos-to-tasmota #20: Shelly Plug (not S and not US version)

yahwehPT, 2020-12-23:
It’s possible to use this method on a shelly plug https://shelly.cloud/products/shelly-plug-smart-home-automation-device/ ?  Thank you

yaourdt, 2020-12-23:
I'll look into it right after fixing the update problem with RGBW2, but I think it should be possible

yaourdt, 2020-12-27:
There are two firmware versions, SHPLG-1 and SHPLG2-1. Lets see if it is SHPLG2-1: http://shellyip/ota?url=http://dl.dasker.eu/plug2/firmware.zip

Does that work?

yahwehPT, 2020-12-28:
Something went wrong... the ota worked but then I updated to 9.2.0 and bricked my device.

yaourdt, 2020-12-29:
Ok, just to confirm: You initially had a working version of tasmota? If so, what did you do next?

yahwehPT, 2020-12-29:
Yes it was working but I upgrade to tasmota 9.2.0 minimal and it worked and then I installed tasmota 9.2.0 and nothing. I used the web gui and choose file for the upgrade.

yaourdt, 2020-12-29:
I'm afraid this is more of a Tasmota issue than a problem with this tool 😐 The moment the first version of Tasmota boots, my code is done running. [This wiki page](https://tasmota.github.io/docs/Upgrading/) should give you all the details you need, or you can open an issue at [arendst/tasmota](github.com/arendst/tasmota/).

yahwehPT, 2020-12-29:
True. Will do. Just for future reference this plug has two versions, one with an ESP chip and another with a CC3200 chip. According to a post on the official facebook support page the plug v2 is the ESP chip.

yaourdt, 2020-12-29:
> Just for future reference this plug has two versions, one with an ESP chip and another with a CC3200 chip. According to a post on the official facebook support page the plug v2 is the ESP chip.

This is actually good to know. Thank you, and good luck with the unbricking!

closed 2020-12-29 by yaourdt