yaourdt/mgos-to-tasmota #22: SUGGESTION: Increase tasmota-minimal version to 9.x

schrej-zz, 2020-12-28:
As a lot of the Shelly templates from blackadder are already 9.x versions, I suggest to raise the tasmota-minimal version in your firmware also to 9.x.
If not, and you forget to upgrade to 9.x before utilizing the templates, you might have a messy GPIO configuration.

Thanks again for you excellent work.

yaourdt, 2020-12-28:
I actually just moved from pinned versions to automatically downloading the latest release of each target firmware. We had that discussion a while back, and I decided to pin versions, as it is good practice.

However, @RavenSystem  ended up waiting for me to update his versions quite frequently, and @arendst forked this repo for (I think?) the same reason, so now we'll just trust in them not changing the download links, and download the latest version directly from the respective release pages.

schrej-zz, 2020-12-28:
Hi Mark,
even better, I think.
As long as they do not change the download link (again).

closed 2020-12-28 by schrej-zz