yaourdt/mgos-to-tasmota #23: Shelly Plug S to ESPurna failed

Torxgewinde, 2021-01-12:
Hello,
Thanks for providing this solution. Unfortunately I had issues converting a "Shelly Plug S" to ESPurna. Once connected to WiFi I issued the command "wget http://192.168.179.53/ota?url=http://dl.dasker.eu/firmware/mg2espurna-ShellyPlugS.zip".

The first stage seems to work (ping stopped for ten seconds and then continued), but the process stopped  after that and did not finish for several minutes. 

Current state of the device:
I can still access the webserver but it just gives a directory listing with a couple of files. Amongst them some JSON and the index gzipped tarball. The OTA endpoint is not working anymore (File not found). Pinging the device works.

Is there anything I can do besides opening the device and flashing over-wires? Did I do anything wrong?

Best regards,
Torx

yaourdt, 2021-01-13:
@Torxgewinde thanks for reporting! I'll have to investigate why ESPurna fails at the moment.

Your devices can be recovered. If you are using the latest version of mgos2... you should be able to update the download URL to any binary of your choosing:

```
curl -d '{"config":{"mg2x":{"url":"<yourTargetUrl>"}}, "save": true,"reboot": true,"level": 1}' <shellyIP>/rpc/Config.Set
```
The device will reboot and download the target file.

I did not have time to test the command, so there might be a typo in it. You can for example use `https://ota.tasmota.com/tasmota/release/tasmota.bin` as `<yourTargetUrl>`.

As an alternative, you could update to another version of mg2x by pushing the matching zip as OTA update with

```
curl -i -F filedata=@./fw.zip http://<shellyIP>/update
```
Replace `fw.zip` with the tasdmota version of mg2....

Please let me know if this helped

Torxgewinde, 2021-01-13:
Hallo Mark,
Thanks for your reply, it helped to flash to Tasmota! I am also not sure what happened with ESPurna, but I am happy that I do not have to fiddle with the tiny cables and well hidden screws :)

What did NOT work:
`curl -d '{"config":{"mg2x":{"url":"https://ota.tasmota.com/tasmota/release/tasmota.bin"}}, "save": true, "reboot": true, "level": 1}' http://192.168.179.53/rpc/Config.Set`
That command came back with:
`{"code":-1,"message":"error saving config: STA SSID must be between 1 and 31 chars"}`

I also misread your info to use the mg2x specific ZIP file from dasker and tried this command first with a regular tasmota download:
`curl -i -F filedata=@tasmota.bin http://192.168.179.53/update`
Which is refused with the answer:
~~~~
HTTP/1.1 500 Internal Server Error
Server: Mongoose/6.18
Content-Type: text/plain
Connection: close

-1 Malformed archive (invalid header)
~~~~

But, in the end it did succeed:
`curl http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip -o mg2tasmota.zip`
`curl -i -F filedata=@mg2tasmota.zip http://192.168.179.53/update`
The device answered with:
~~~~
HTTP/1.1 200 OK
Server: Mongoose/6.18
Content-Type: text/plain
Connection: close

1 Update applied, rebooting
~~~~

Et voila, after a while a new SSID was broadcasted and I had it converted. Great! Flashing from Tasmota to ESPurna is possible with `SetOption78 1`, so I am not stuck with Tasmota in case I do not like it too much. The MQTT strings are IMHO strange, but certainly that is a matter of taste.

Thanks a million for your help and especially for your project!

yaourdt, 2021-01-14:
Ha! Nice!

For the first approach: If you still have a spare, try it with `level=9`, if not, it doesn't matter too much.

The malformed header error is to be expected, as you cannot update Mongoose OS directly using a bin file due to different update mechanisms. This is the reason this project exists :)

For the second approach: Glad it worked and saved you the hassle of opening your devices. I'll still leave this issue open for now, so I remember to fix the original problem

Torxgewinde, 2021-01-15:
Yes, I will try with the remaining Shelly 1L if `level=9` will work and come back to you.

radven, 2021-01-15:
I just ran into this same problem trying to install HomeKit Accessory Architect via mg2haa-Shelly1.zip

I haven't tried the advice you give above to recover yet.

What is the root problem that gets the Shelly hung up in the intermediate phase?

Cheers,
  - Chris

yaourdt, 2021-01-18:
I think I have found the issue. I'll give you more details later today

yaourdt, 2021-01-18:
Ok, let me give you a more detailed explanation: Starting from version 0.4.3 this firmware downloads the latest target releases of tasmota, haa, espurna directly from the respective release pages. In case of haa and espurna, this means following 302 redirects. My implementation for following those redirects was not 100% compliant with HTTP standards, which resulted in the intermediate firmware sometimes not following the redirect.

I've fixed this with v0.4.6, which is currently building. For new devices, you should not get stuck any longer. Devices with older firmware versions, that are stuck in the process, can be recovered using the latest intermediate firmware. Please download the latest matching release and apply it using

```
curl -i -F filedata=@./fw.zip http://<shellyIP>/update
```

Your device should then continue to the desired target firmware. @radven can you please confirm?

radven, 2021-01-18:
I tried 0.4.6, and left if sit for 15 minutes and HAA never installed. It seems to be behaving exactly the same as before.
```
chris@Chriss-MacBook-Air Downloads % curl -i -F filedata=@mg2haa-Shelly1.zip http://192.168.50.106/update
HTTP/1.1 200 OK
Server: Mongoose/6.18
Content-Type: text/plain
Connection: close

1 Update applied, rebooting
```
Let me know if there is anything else I should try for you.

  - Chris

yaourdt, 2021-01-18:
@radven - we had this in other issues before, so please double-check:
 1. Your device is in a network with automatically assigned IP addresses, has the right subnet mask, DNS server, and router configuration?
 2. Your device is allowed to reach the internet?

I think you can use `curl http://<shellyIP>/rpc/Sys.GetInfo` to confirm these settings.

Also, please set the download URL to `https://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin` instead of `http://...`, as the redirect from `http` to `https` is 301, which is not implemented.

radven, 2021-01-18:
> 1. Your device is in a network with automatically assigned IP addresses, has the right subnet mask, DNS server, and router configuration?
> 2. Your device is allowed to reach the internet?
> 
> I think you can use `curl http://<shellyIP>/rpc/Sys.GetInfo` to confirm these settings.

This is what I see:
```
chris@Chriss-MacBook-Air Downloads % curl http://192.168.50.106/rpc/Sys.GetInfo
{"id": "shelly1-8CAAB5C51510", "app": "switch1", "fw_version": "0.4.6", "fw_id": "20210118-130045", "mg_version": "2.19.0", "mg_id": "20210118-130045/2.19.0-ge456939", "mac": "8EAAB5C51510", "arch": "esp8266", "uptime": 25, "public_key": null, "ram_size": 52936, "ram_free": 40876, "ram_min_free": 38768, "fs_size": 233681, "fs_free": 119727,"wifi": {"sta_ip": "192.168.50.106", "ap_ip": "", "status": "got ip", "ssid": "Y-Not Bytes"}}
```

I am guessing by the missing ap_ip the Shelly is not getting configured properly?

But on the router - I see the DHCP reservation and the 192.168.50.106 address being assigned. There is no reason that the Shelly should be blocked from the internet - no other DHCP devices are having any issues.

> Also, please set the download URL to `https://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin` instead of `http://...`, as the redirect from `http` to `https` is 301, which is not implemented.

I just tried this:
```
chris@Chriss-MacBook-Air Downloads % curl -d '{"config":{"mg2x":{"url":"http://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin"}}, "save": true,"reboot": true,"level": 9}' http://192.168.50.106/rpc/Config.Set
{"saved": true}
```
Still the same.

Ideas?

yaourdt, 2021-01-19:
It is ok for `ap_ip` to be empty. 

Yes, please replace `http` with `https` in your download URL, as I suggested above:

```
curl -d '{"config":{"mg2x":{"url":"https://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin"}}, "save": true,"reboot": true,"level": 9}' http://192.168.50.106/rpc/Config.Set
```

radven, 2021-01-19:
> Yes, please replace `http` with `https` in your download URL, as I suggested above

For some reason the first try with https didn't work, so I tried http. But when I switched back to https, it worked great - running HAA now. Problem solved!

Thanks!  :-)

yaourdt, 2021-01-19:
Nice! Thanks for sticking with me, and I'm glad it worked \o/

closed 2021-01-19 by yaourdt

nimal, 2022-08-02:
Hi,-

I just ran into the same problem as @Torxgewinde with a Shelly Plug S. I tried to follow all the steps described above, but unfortunately without success.

When I use the command:

`curl -i -F filedata=@mg2tasmota.zip http://x.x.x.x/update` 

The device is not getting updated. However, I can see something is actually going on. The updater.dat on the webserver contains the following:

`HTTP/1.1 409 OK
Server: Mongoose/6.18
Content-Type: text/plain
Connection: close

Another update is in progress.`

But Curl command never returns Update applied.

I also tried the command:

`$ curl -d '{"config":{"mg2x":{"url":"https://ota.tasmota.com/tasmota/release/tasmota.bin"}}, "save": true,"reboot": true,"level": 9}' http://x.x.x.x/rpc/Config.Set`

I can see the mg2x config is updated with the url, but the device is not downloading the target file.

`ll_up":true},"btn2":{"pin":-1,"pull_up":false},"btn3":{"pin":-1,"pull_up":false}},"mg2x":{"url":"https://ota.tasmota.com/tasmota/release/tasmota.bin"}}
`

Any help is appreciated.

[image]