yaourdt/mgos-to-tasmota #24: Shelly Duo - RGBW GU10 - New Device

holdestmade, 2021-01-14:
Any chance we can add this device ?

I think the firmware is:

http:\\shelly-api-eu.shelly.cloud\firmware\SHCB-1.zip

Thanks

```
    "SHCB-1": {
      "url": "http:\/\/shelly-api-eu.shelly.cloud\/firmware\/SHCB-1.zip",
      "version": "20210107-122133\/1.9_GU10_RGBW@07531e29"
    },
```

yaourdt, 2021-01-14:
I'll look into it within the next days 👍

yaourdt, 2021-01-14:
Ok, please try `http://shellyip/ota?url=http://dl.dasker.eu/shellyduorgbwtest/firmware.zip` and confirm - it might work, but no promises. Could be that we need to treat it like the Shelly RGBW, we need to try that

yaourdt, 2021-01-14:
A meeting got cancelled, ended up with more time than expected today :)

Cool that it works on the first try! I'll leave the issue open until I release it to the main branch

holdestmade, 2021-01-14:
Its not working correctly though, it only works as a toggle - red off, green on, no colour controls

yaourdt, 2021-01-14:
Ah, now I get it. Well, that problem is beyond the scope of this project, sorry. As soon as Tasmota starts, this code is done. If you find no matching template, you'll either need to [build it yourself](https://tasmota.github.io/docs/Templates/), or get in touch with the people maintaining Tasmota

holdestmade, 2021-01-14:
OK, have sorted it by trial and error now, thanks for doing the firmware

yaourdt, 2021-01-14:
You are welcome :)

Would you mind sharing the template with me, so I can add it to the README? Also, please consider posting it on templates.blakadder.com, so others can profit from it

holdestmade, 2021-01-14:
I'm new to Tasmota and I couldn't figure out how to make it show up as an RGBW light, only separate channels.

I use esphome so uploaded a firmware from there and it works great.

I've forgot what template got me going but the GPIOs used are:

GPIO4 - White
GPIO5 - White
GPIO12 - Green
GPIO13 - Red
GPIO14 - Blue

yaourdt, 2021-01-18:
Current release supports the Shelly Duo RGBW, so I'll close this. Thank you @holdestmade !

closed 2021-01-18 by yaourdt

PeterLeipe, 2021-09-27:
What template did you use? The last one you send @holdestmade does not work for me. I would appreciate it, if you can look it up.
Thanks in advance. Peter