yaourdt/mgos-to-tasmota #25: Shelly Plug S to Tasmota failure

CamielFolmer, 2021-01-19:
Hi,

Firs of all thanks for your work making this possible. I just encountered an issue with flashing a Shelly Plug S with tasmota.

I've connected the plug WiFi, and then I issued the command : http://<shelly Plug S IP>/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip 

I then saw: 
{"status":"updating","has_update":false,"new_version":"20210115-103659/v1.9.4@e2732e05","old_version":"20190516-073020/master@ea1b23db"}

The leds on the plug started flashing but stopped after a while, I waited for about ten minutes, then replugged the plug but nothing, it doesn't connect to WiFi or creates an SSID. Also the rings on the led are not flashing. There's only a led on the board (D2) solid red.. (not the ring)

I did exactly the same thing about a week ago with success. the only thing is see in the outut when flashing is new_version":"20210115" I did the succesful flash before 15-01 maybe there's a difference?

Any advise I can do? I'm already looking how to physically connect to the plug to flash it that way.

yaourdt, 2021-01-19:
If you can connect to the plug directly, logging output would be much appreciated. 

I'm not sure yet in which state the plug is. Can you ping it? Can you read the status, e.g. by using `curl http://<shellyIP>/rpc/Sys.GetInfo`?

CamielFolmer, 2021-01-21:
The plug was not connecting to a network or creating a wirelless network. So I now flashed it using an usb cable.. I will try the next plug using he same procedure again, knowing I can recover by connecting it to USB. Will let ou know

yaourdt, 2021-01-21:
Ok, next time, if you have USB attached anyway, can you please post the device logs? So far, we just know there is a bug, we don't know where exactly the problem is, so I can't fix it... you would be doing me (and others that might benefit from a bug fix) a huge favour :)

CamielFolmer, 2021-01-21:
Yes, Sorry I wanted to do that, but had connecting issues and I pressed flash rather then view logs.. ill try the next plug in a few hours.

yaourdt, 2021-01-21:
Ok, let me know how it goes 😊

CamielFolmer, 2021-01-21:
Just did a new plug and the tasmota firmware was succesfully flashed! So not sure what went wrong with the previous one. Thanks, and sorry for bothering you :)

yaourdt, 2021-01-21:
No worries! Maybe it was a network problem. The firmware cannot check the download for integrity, as the firmware maintainers don't publish hashes. I'll close this for now, but I'll keep it in mind to try and fix this problem in a future release

closed 2021-01-21 by yaourdt

daloki83, 2022-10-19:
Hi, currently I’m having the same problem with the EU version. Flashed 12/13 Shelly’s successfully but one failed with the exact same problem as described above. Tomorrow i will receive my USB2TTL converter and will flash it directly with the Tasmota-Flashtool. So far i couldn’t see a option for logging. 
If I can support you, just tell me how. 
[image]