yaourdt/mgos-to-tasmota #26: Update HAA Installer to v4.3.1

RavenSystem, 2021-01-23:


yaourdt, 2021-01-24:
Thanks for the heads up! Starting with v0.4.0 I've switched to downloading the latest target firmware directly from the maintainers, in your case `https://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin`. If you ever happen to change that URL, please let me know!

Here is some context: https://github.com/yaourdt/mgos-to-tasmota/issues/22#issuecomment-751722759

Thanks by the way for your awesome firmware!

RavenSystem, 2021-01-24:
Ohh, I see. Many thanks!! Yes, Tasmota developer created a fork to use directly from his repo. Don't worry about URL change, if It changes, there will be a big mess. 

I personally don't try your firmware because I use wires to flash (it is faster for me), but many users use your firmware to install mine and it works great. Thanks again!!

closed 2021-01-24 by RavenSystem