yaourdt/mgos-to-tasmota #27: Flashing other firmware

ivanvach, 2021-01-29:
Thanks for the great tool. Wondering if I can flash any .bin file. I'm using zumungo.com ESP software and  I like everything to be on the same platform. For example will this work

http://shellyip/ota?url=http://myownwebserver/htdocs/mybinfile.bin

 , where mybinfile.bin is the binary file that I want to flash into Shelly. Typically it comes from Arduino IDE compiler - "Export compiled  binary" but in the case of zumungo.com it comes from the vendor per-compiled .bin file.

yaourdt, 2021-01-29:
The firmware is not yet designed to accept arbitrary URLs, however, you can work around that. This would be the procedure (but be aware it is untested and requires some background knowledge with regard to networking, so proceed at your own risk):

Variant A:

1. Compile your own version of this firmware, setting the download URL in `mos.yml` to whatever target bin file you want
2. Host the resulting fw.zip on a web server of your choosing, which can serve files over HTTP (not HTTPS due to limitations of the stock firmware!)
3. Update with `http://shellyip/ota?url=http://yourownwebserver/fw.zip`

Variant B:
1.  Cut the device's internet access and host a matching build of this firmware on a **local** web server of your choosing, which can serve files over HTTP (not HTTPS due to limitations of the stock firmware!),
2. Update with `http://shellyip/ota?url=http://yourownlocalwebserver/fw.zip`
3. Due to the missing internet access, the firmware will not be able to proceed to tasmota. Alter the download URL using this command: `curl -d '{"config":{"mg2x":{"url":"https://yourtarget.bin/file.bin"}}, "save": true,"reboot": true,"level": 9}' http://shellyip/rpc/Config.Set`
4. Re-enable internet access

If you try it, let me know how it goes. If I have some spare time during the next months I'm planning on releasing a version one can use to flash any bin file, until then, these are the ways

ivanvach, 2021-01-30:
Thanks for the prompt and detailed response. Tried both no luck.
Plan A. 
There is no setting for "download URL" in mos.yml file. There are "Libraries" but I think that it is something else. Also I couldn't find any documentation on the subject that would help adding "download URL".
Plan B
I can't put a file fw.zip  on my server "http://yourownlocalwebserver/fw.zip"  that I don't have (p.2. of plan B)

NdS-Research-Facilities, 2021-01-31:
> Plan B
> I can't put a file fw.zip on my server "http://yourownlocalwebserver/fw.zip" that I don't have (p.2. of plan B)

I believe what Mark meant to say was that you should use the matching firmware update url for your device from this site:
https://github.com/yaourdt/mgos-to-tasmota

ivanvach, 2021-01-31:
This makes sense but than in p.3 of Plan B same issue as in plan A. How to alter "download URL" when there is no "download URL" in the mos.yml file to begin with.

yaourdt, 2021-01-31:
Hi @ivanvach - Variant A: You should use the [mos.yml](https://github.com/yaourdt/mgos-to-tasmota/blob/main/mos.yml) and other files from this Repo, not from the Mongoose demo - `git clone` this repository, replace mg2x.url, and follow the [instructions at the bottom of the README.md file](https://github.com/yaourdt/mgos-to-tasmota#build-the-firmware-yourself). 

Variant B: You need to host a copy of Mongoose to Tasmota locally and cut the internet access to prevent the device from downloading the target firmware during the process, so you have time to modify the download URL. As I said: It requires some background knowledge with regard to networking, e.g. setting up your own local web server.

If you are not sure about how to do this, I strongly recommend waiting until I or someone else gets around to finishing this feature (but this may take a while).

ivanvach, 2021-01-31:
Hi Mark, thanks for the hand holding. Making progress now better understanding the process but still no luck.

1. I'm getting errors from mos when cloning from your repository (see screenshot below).
2. Regardless of the errors mos does create directory "mgos-to-tasmota" with all of your files incl. mos.yml file that has all of your stuff. Here I did find "download URL" clearly marked, which solves previously reported problem. I did replace the URL with the URL that contains my .bin file. No problem.
3. Unfortunately "mos build" command failed (see errors in the screenshot below) and I'm still not able to get the coveted fw.zip file
[image]

yaourdt, 2021-02-01:
Can you please post the Shelly 1 device logs?

ivanvach, 2021-02-01:
Here are the device logs from Shelly. Yes indeed something was installed and working but it doesn't create AP as it normally would. Here is the log 

07:09:39.520 -> Started!
07:09:39.520 -> Clear memory!
07:09:40.059 -> Memory cleared
07:09:40.059 -> WiFi SSID not set
07:09:40.059 -> http://www.zumungo.com/wp-content/uploads/firmware/ProjectR1.bin
07:09:40.059 -> Begin OTA process
07:09:40.206 -> HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused
07:09:40.206 -> OTA finished
07:09:40.206 -> ESP restarts...
07:09:41.304 -> 
07:09:41.304 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,7)
07:09:41.304 -> 
07:09:41.304 -> load 0x4010f000, len 1392, room 16 
07:09:41.351 -> tail 0
07:09:41.351 -> chksum 0xd0
07:09:41.351 -> csum 0xd0
07:09:41.351 -> v3d128e5c
07:09:41.351 -> ~ld
07:09:41.399 ->

closed 2021-02-01 by ivanvach