yaourdt/mgos-to-tasmota #28: Shelly Plug US flashing failed

kbaker180, 2021-01-29:
I had previously flashed 2 Shelly US plugs without issue, back in late October, and got around to flashing my last 2 tonight, and both of them have failed.  I followed the same procedure as I previously did, but this time around they never came up with their own wifi network, and are not sitting on any accessible IP that I can find.   

I have tried to reset them with the fast power cycle recovery, and still no luck.   Seems odd that what worked in October is no longer working.  One plug had been used for the few months with the Shelly firmware, the other sat unused for that time.  Any recommendations?

yaourdt, 2021-01-29:
It is odd indeed, I'll take a look. Did you try to flash Tasmota, ESPurna, or HAA?

kbaker180, 2021-01-29:
Tasmota with this link: http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugUS.zip.  I didn’t really pay attention to them during the flash because it worked so well on the first two.  I do remember seeing one blink red/blue/red/blue quickly right after issuing the command.  A few minutes later it rebooted and now both have just a dimmer red light.

yaourdt, 2021-02-01:
I just tried to reproduce the issue, but for me, it worked without problems. I think this is the same problem as #25 (and maybe #27 ) - can you access the device logs?

kbaker180, 2021-02-02:
I can't access the devices at all.  If I have time I may try and crack them open and get to the pins to flash them that way.  I've not been able to recover them at all (holding button or fast power cycle method).

yaourdt, 2021-02-02:
I'm sorry to hear. If you decide to open the devices, logs and a flash dump would be very helpful to understand the underlying issue.

jimmypuckett, 2021-02-21:
I am now having this issue too.  I have flashed about 15 of the 2.5 over the past month or so, but I ran out & had to order more.  They came in last week & I installed one put the cover back on & later that night flashed it.  It says that it all worked out, but I never get the tasmota wifi to join.  I had to pull it out & flash it with the pins.  I just tried as 2nd one, and it is doing the exact same thing.

@yaourdt what logs can I get you to see if we can trace this out?  I have 3 to flash & 8 more one the way.

jimmypuckett, 2021-02-21:
BTW - the firmware on this batch of 6 is `20191217-074901`.  There is `20210115-103025` available, so I think that I am going to update to that version & then try flashing on a different one?

jimmypuckett, 2021-02-21:
OK, I updated another one to `` & then did the flash.  I got back this json

```json
{
   "status":"updating",
   "has_update":false,
   "new_version":"20210115-103025/v1.9.4@e2732e05",
   "old_version":"20210115-103025/v1.9.4@e2732e05",
   "beta_version":"20210209-101632/v1.10.0-rc2-g0cf9ee3-heads/v1.10.0-rc2"
}
```

tomajask, 2021-02-22:
Interesting... I'm having the same issue but with Shelly Plug S. I have 2 of those. Flashing the first went well, it works. The second failed. It's not connected to the WiFI and also I cannot connect to Tasmota's WiFi 😕 
I tried recoevery options listed here: https://tasmota.github.io/docs/Device-Recovery/ but with no luck.

When plugging it to the socket it blinks once with blue light and that's it. It's a brick 😞

deepcoder, 2021-04-24:
I am trying your procedure on a set of Shelly Plus US units.  1st was success, 2nd failed (it will not even upgrade to newer Shelly firmware), 3rd success. I have 4 more to go, fingers crossed. Shelly has got a real problem with something, hardware or firmware. Several of my plugs started to randomly turn off for a 1 second and turn back on, this starting back in November 2020. I did not do a great job of recording my firmware upgrades, so I cannot give exact versions of firmware where the problem started. But of the three devices I have upgraded so far, unit 1 was on: v1.10.3-g23074d0, unit 2 (the one that will not upgrade or convert) is on: 20200827-070340/v1.8.3@4a8bc427 , unit 3 that did convert was on: 20201007-085117/v1.8.5@5fbcbaa3

Thank you @yaourdt for your work on this conversion, much appreciated! I was a fan of the Shelly devices, but no more. I really feel for the folks that are all in on these devices if they start to have the problems I was seeing. I had one Shelly on a refrigerator when it started the power drops, I suspect that if I had not caught it, it would have damaged the fridge.

I'm happy to send you the failed Shelly Plug US (and any others that am not able to convert, if you want them.

OldButEager, 2022-10-09:
I have 6 ShellyPlug (plus) US and none of them will flash Tasmota
http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugUS.zip 
yes, i changed it to the proper ip.
Worked perfectly on 10 prior ShellyPlug US (not plus).