yaourdt/mgos-to-tasmota #37: Shelly 1 v3 OTA updated did not worked

manurosado, 2021-03-30:
I used URL => http://xx.xx.xx.51/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-Shelly1.zip  on my Shelly 1 v3 and it never came back, I got the confirmation that it was going to do the update, waited 10 minutes and then no Tasmota wifi, neither could not get back from the IP it had, in which it was a static IP

kerkmann, 2021-06-11:
Can confirm that. :)\
Same thing happened on my shelly 1 v3.

stefanheijnen, 2021-07-04:
Same here. I tried two Shelly 1 v3's. 1 failed, 2nd succeeded.

leandrotoledo, 2021-08-13:
This has also happened to me.

Using Shelly 1 v3, originally set up in the Android app.

`{"status":"updating","has_update":false,"new_version":"20210720-184522/v1.11.0-g6abd92e","old_version":"20191217-140757/1047-long-id-for-shelly-devices@a0661583"}`

leandrotoledo, 2021-08-22:
I gave this tool another try with another new Shelly 1 I had laying around and it was successful.

The only difference here was that I upgrade the Shelly 1 firmware to the latest available before upgrading to Tasmota.

*Before:*
```
Current version: 20210720-184522/v1.11.0-g6abd92e
You have latest version on your device!
```

*During:*
```
{"status":"updating","has_update":false,"new_version":"20210720-184522/v1.11.0-g6abd92e","old_version":"20210720-184522/v1.11.0-g6abd92e","beta_version":"20210812-134038/v1.11.4-rc1-g03ae3f6"}
```

*After:*
```
Program Version | 9.5.0(tasmota)
```

I'm not sure this is the definitive fix here, but may be a lead. It needs further testing.

schenkl, 2023-11-26:
I have the same issue.  I have successfully flashed a dozen Shelly 1's, this time nothing happened.  They kept the address I had set them to, tried the OTA Tasmota flash, nothing happened.  So I went back to hooking up the wires and flashing from the tasmota web page.  It worked the first try.  Has anyone figured out what is going on?  As least I didn't brick them!