yaourdt/mgos-to-tasmota #38: Shelly Duo - never see tasmota access point

transfinite8, 2021-04-08:
I just got a Shelly Duo bulb. After following the instructions, and the video, I used
    `http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyDuo.zip`
It returned
```
status	"updating"
has_update	false
new_version	""
old_version	"20191216-140245/???"
```
But I never saw the tasmota access point, and now I'm getting nothing from the bulb. I've turned it on and off multiple time since with no luck.

Is my only option left to attempt to open the bulb and solder wires to it? Is there anything else I should try?

eignatov, 2021-05-28:
Hello. I have same issue. Did you resolve this issue?

stehof2, 2021-09-11:
> Hello. I have same issue. Did you resolve this issue?

And I’m having the same issue, did you manage to fix it?

BertoldVdb, 2022-11-09:
I had the same result. The bulb was flashing for a while and then rebooted to black. Now it does nothing. Will try to get to the serial port and investigate.

rammelsberg, 2023-01-04:
Hello!
Exactly the same for me. First blinking white while updating then going off. No response for a couple of minutes neither when cycling power.
I opened the bulb (just pulled at the transparent dome part) and disassembled the led board. Below is a power board with a mounted (soldered) esp8266 module.
It has two rows of 8 connection pads, looks like an esp07.
To the led board there is an 8 pin connector labeled with vcc, gnd, 24v,r,g,b,cw,ww what i assume is power for logic and leds + three colors and 2 extra white colors (warm and cold)
The gnd has a connection to the modules bottom right which is gnd for esp07, would fit therefore. The vcc i was not able to find at the module neither the 2 white connections. The color connections are at gpio 12,14 and 4.
At the module yhere is also a nor flash in so8 package. The gnd and vcc connections of that flash have a connection to the right and left bottom connection pad of the module, that would also fit an esp07 module.
Maybe tomorrow i will give it a try and connect my serial programmer, i hope if works...
Greetings!

ivanpavlina, 2023-01-30:
@rammelsberg did you manage to resolve this, i have exact same problem

rammelsberg, 2023-01-30:
Hi!
I did solve this, i dont know the exact module type but the connection is like the tywe3l pinout for gnd vcc and rx tx.
Connecting the serial programmer and running tasmotizer did the trick at last.
I then just glued the glass dome to the rest with some silicone.
The lamp works perfect now...
Greetings,
Paul

ivanpavlina, 2023-01-31:
> Hi! I did solve this, i dont know the exact module type but the connection is like the tywe3l pinout for gnd vcc and rx tx. Connecting the serial programmer and running tasmotizer did the trick at [...]

Great, thank you!

praguevara, 2023-05-11:
@rammelsberg hey I've just run into the same problem. Do you mind going into some more detail about the flashing process? Which are the pins I need to connect to? Sorry I'm not very experienced with this. Thanks