yaourdt/mgos-to-tasmota #40: Warning, will brick your device.

jimmyw, 2021-04-18:
Bricked 2 Plug S.

One i had updated to latest firmware, one was not toched before i tried.

Did follow instructions, after ota device rebooted keeping its ip and you could se a filesystem if you visited the old address.

After quite a long time, 15 minutes or so the device rebooted.

No tasmota, did attach serial, and can se a repeating message sent, but could not decode the baudrate.

Tried 115200, 57600, 74880 could not decode anything.

Pulled GPIO 0 low, got esptool.py working, but flashing tasmota failed.

jimmyw, 2021-04-29:
Got some serial output:

```
rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   16 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0 (0x8000).
add i
bcn 0
del if1
usl
mode : null
mode : sta(e0:98:06:b5:7d:8d)
add if0
sleep disable
WPA2 ENTERPRISE VERSION: [v2.0] disable

SHPLG-S-B57D8D E0:98:06:B5:7D:8D

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
pm open,type:0 0
state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with JNET, channel 11
dhcp client start...
ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1
pm open,type:0 0
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : null

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1884, room 16 
tail 12
chksum 0x4b
ho 0 tail 12 room 4
load 0x3ffe8000, len 824, room 12 
tail 12
chksum 0x24
csum 0x24?????.@k??%x?t.v?~?k?(q????j?(v ??JEN(|?xJ~?-(?c #?-A%v?-.??x ??+P?n???Xvj|?? ??cPz~?h?n?,PJx~????kAj?????n?+?kA,%,R???8?V2
esp_main.c:165          shelly-plug-s 0.4.6 (20210118-130847)
esp_main.c:167          Mongoose OS 2.19.0 (20210118-130847/2.19.0-ge456939) sp 0x3ffffe80
esp_main.c:169          CPU: ESP8266EX, 160 MHz, RAM: 52936 total, 49852 free
esp_main.c:173          SDK 3.0.5-dev(52383f9); flash: 2M
esp_exc.c:235           Reset cause: 4 (soft reset)
mongoose.c:15676        Mongoose 6.18, LwIP 2.1.2
[... 63 lines cut ...]
main.c:261              device is online, downloading target firmware
mgos_mongoose.c:66      New heap free LWM: 36872
mgos_mongoose.c:66      New heap free LWM: 36352
mongoose.c:4912         0x3fff0994 ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
mgos_mongoose.c:66      New heap free LWM: 31360
mgos_mongoose.c:66      New heap free LWM: 30632
mgos_mongoose.c:66      New heap free LWM: 29224
esp_main.c:145          SDK: pm open,type:0 0
mgos_mongoose.c:66      New heap free LWM: 23088
mgos_mongoose.c:66      New heap free LWM: 18864
mgos_mongoose.c:66      New heap free LWM: 18712
mgos_mongoose.c:66      New heap free LWM: 18664
mgos_mongoose.c:66      New heap free LWM: 17696
mgos_mongoose.c:66      New heap free LWM: 17304
mongoose.c:4912         0x3fff0994 mbedtls_ssl_flush_output() returned -78 (-0x004e)
mongoose.c:4912         0x3fff0994 mbedtls_ssl_write_record() returned -78 (-0x004e)
mongoose.c:4912         0x3fff0994 mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
main.c:162              HTTP status is 200, recieved 0 bytes

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
ets_main.c r

```

jimmyw, 2021-04-29:
Again, mgos-to-tasmota recompiled with debug flags:
`esp_main.c:145          SDK: connected with JNET, channel 6
esp_main.c:145          SDK: dhcp client start...
mgos_mongoose.c:66      New heap free LWM: 45448
mgos_wifi.c:133         WiFi STA: Connected, BSSID 68:d7:9a:45:ce:aa ch 6 RSSI -64
mgos_event.c:134        ev WFI2 triggered 0 handlers
mgos_net.c:90           WiFi STA: connected
mgos_event.c:134        ev NET2 triggered 2 handlers
esp_main.c:145          SDK: ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1
mgos_mongoose.c:66      New heap free LWM: 45264
mgos_event.c:134        ev WFI3 triggered 0 handlers
mgos_net.c:101          WiFi STA: ready, IP 192.168.1.126, GW 192.168.1.1, DNS 192.168.1.1
main.c:261              device is online, downloading target firmware
main.c:253              fetching https://ota.tasmota.com/tasmota/release/tasmota.bin to 0x8000
mongoose.c:8604         https://ota.tasmota.com/tasmota/release/tasmota.bin use_ssl? 1 tcp://ota.tasmota.com:443
mongoose.c:3142         0x3fff096c tcp://ota.tasmota.com:443 -,-,*
mongoose.c:3142         0x3fff2964 udp://192.168.1.1:53 -,-,-
mongoose.c:3012         0x3fff2964 udp://192.168.1.1:53
mgos_event.c:134        ev NET3 triggered 2 handlers
mongoose.c:3026         0x3fff2964 udp://192.168.1.1:53 -> 0
mgos_mongoose.c:66      New heap free LWM: 37160
mongoose.c:3012         0x3fff096c tcp://65.9.43.127:443
mgos_mongoose.c:66      New heap free LWM: 36768
mongoose.c:3026         0x3fff096c tcp://65.9.43.127:443 -> 0
mongoose.c:4912         0x3fff096c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
mgos_mongoose.c:66      New heap free LWM: 31776
esp_main.c:145          SDK: pm open,type:0 0
mgos_mongoose.c:66      New heap free LWM: 24152
mgos_mongoose.c:66      New heap free LWM: 19776
mongoose.c:4912         0x3fff096c mbedtls_ssl_flush_output() returned -78 (-0x004e)
mongoose.c:4912         0x3fff096c mbedtls_ssl_write_record() returned -78 (-0x004e)
mongoose.c:4912         0x3fff096c mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
main.c:162              HTTP status is 200, recieved 0 bytes
main.c:174              last block dump done
main.c:33               block_copy start: cp 4096 bytes from 0x8000 to 0x9000
main.c:75               block_copy done
main.c:33               block_copy start: cp 4096 bytes from 0xf000 to 0xa000
main.c:75               block_copy done
main.c:33               block_copy start: cp 24576 bytes from 0x9000 to 0x1000
main.c:75               block_copy done
main.c:33               block_copy start: cp 4294934528 bytes from 0x10000 to 0x8000

 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
ets_main.c`

jimmyw, 2021-05-01:
```
esp_main.c:145          SDK: aid 8
esp_main.c:145          SDK: cnt
esp_main.c:145          SDK:
esp_main.c:145          SDK: connected with JNET, channel 11
esp_main.c:145          SDK: dhcp client start...
mgos_wifi.c:133         WiFi STA: Connected, BSSID 68:d7:9a:45:c5:22 ch 11 RSSI -49
mgos_event.c:134        ev WFI2 triggered 0 handlers
mgos_net.c:90           WiFi STA: connected
mgos_event.c:134        ev NET2 triggered 2 handlers
esp_main.c:145          SDK: ip:192.168.1.126,mask:255.255.255.0,gw:192.168.1.1
mgos_event.c:134        ev WFI3 triggered 0 handlers
mgos_net.c:101          WiFi STA: ready, IP 192.168.1.126, GW 192.168.1.1, DNS 192.168.1.1
main.c:266              device is online, downloading target firmware
main.c:258              fetching https://ota.tasmota.com/tasmota/release/tasmota.bin to 0x8000
mongoose.c:8604         https://ota.tasmota.com/tasmota/release/tasmota.bin use_ssl? 1 tcp://ota.tasmota.com:443
mongoose.c:3142         0x3fff0e6c tcp://ota.tasmota.com:443 -,-,*
mongoose.c:3142         0x3fff0f84 udp://192.168.1.1:53 -,-,-
mongoose.c:3012         0x3fff0f84 udp://192.168.1.1:53
mgos_event.c:134        ev NET3 triggered 2 handlers
mongoose.c:3026         0x3fff0f84 udp://192.168.1.1:53 -> 0
mgos_mongoose.c:66      New heap free LWM: 36728
mongoose.c:3012         0x3fff0e6c tcp://65.9.57.86:443
mgos_mongoose.c:66      New heap free LWM: 36336
mongoose.c:3026         0x3fff0e6c tcp://65.9.57.86:443 -> 0
mongoose.c:4912         0x3fff0e6c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
mgos_mongoose.c:66      New heap free LWM: 31344
esp_main.c:145          SDK: pm open,type:0 0
mgos_mongoose.c:66      New heap free LWM: 23720
main.c:96               Connected to ota server.
main.c:106              Got chunk size: 0 received: 0
mgos_mongoose.c:66      New heap free LWM: 19344
mongoose.c:4912         0x3fff0e6c mbedtls_ssl_flush_output() returned -78 (-0x004e)
mongoose.c:4912         0x3fff0e6c mbedtls_ssl_write_record() returned -78 (-0x004e)
mongoose.c:4912         0x3fff0e6c mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
main.c:106              Got chunk size: 0 received: 0
main.c:167              HTTP status is 200, recieved 0 bytes
main.c:179              last block dump done
main.c:33               block_copy start: cp 4096 bytes from 0x8000 to 0x9000
main.c:75               block_copy done
main.c:33               block_copy start: cp 4096 bytes from 0xf000 to 0xa000
main.c:75               block_copy done
main.c:33               block_copy start: cp 24576 bytes from 0x9000 to 0x1000
main.c:75               block_copy done
main.c:33               block_copy start: cp 4294934528 bytes from 0x10000 to 0x8000

 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
ets_main.c
```

jimmyw, 2021-05-01:
Another try now, with a http server to download imafe from.

I get progress, but its slow. 500 bytes a minute until the webserver times out.

Any idea?
```
esp_main.c:165          shelly-plug-s 0.4.6 (20210501-230830)                                   �
esp_main.c:167          Mongoose OS 2.19.1 (20210501-230830/2.19.1-gd52032f) sp 0x3ffffe80       ��p�|����x��ǒ��p
esp_main.c:169          CPU: ESP8266EX, 160 MHz, RAM: 52936 total, 49852 free
esp_main.c:173          SDK 3.0.5-dev(52383f9); flash: 2M                                                        �nn��;�n����
esp_exc.c:235           Reset cause: 6 (sys reset)                                                                           �
mongoose.c:15676        Mongoose 6.18, LwIP 2.1.2                                                                             b�$rrp�n��
mongoose.c:4938         mbed TLS 2.16.6-cesanta1                                                                                        ����l�
mgos_vfs_dev.c:72       sfl0: sysflash (), size 2097152                                                                                       �
mgos_vfs_dev.c:72       root: part ({"dev": "sfl0", "offset": 1814528, "size": 262144}), size 262144                                           �
mgos_vfs.c:173          /: SPIFFS @ root, opts {"bs":4096,"ps":256,"es":4096}                                                                   b�n��n�$�
mgos_vfs.c:344          /: size 233681, used: 97388, free: 136293                                                                                        ��
mgos_sys_config.c:232   Loading conf9.jsonmgos_config_util.c:110  Extra key: [dns_sd]
mgos_config_util.c:110  Extra key: [dns_sd.host_name]
mgos_config_util.c:110  Extra key: [dns_sd]�
mgos_config_util.c:110  Extra key: [mqtt]   �
mgos_config_util.c:110  Extra key: [mqtt.will_topic]
mgos_config_util.c:110  Extra key: [mqtt]     p��<��
mgos_config_util.c:110  Extra key: [first_boot]     ��
mgos_config_util.c:110  Extra key: [timezone]         b��<~�n���
mgos_config_util.c:110  Extra key: [latitude]                   r��#�n�$�l`8`rn|��n��r��l`8���;��
mgos_config_util.c:110  Extra key: [longitude]                                                   ���
mgos_config_util.c:110  Extra key: [tz_utc_offset]
mgos_config_util.c:110  Extra key: [tz_next_dst_shift]
mgos_config_util.c:110  Extra key: [tz_next_utc_offset]
mgos_config_util.c:110  Extra key: [tz_has_dst]
mgos_config_util.c:110  Extra key: [tz_is_dst]
mgos_config_util.c:110  Extra key: [wrong_sta_protect]
mgos_config_util.c:110  Extra key: [wrong_sta_protect.sta_enable]
mgos_config_util.c:110  Extra key: [wrong_sta_protect.sta_ssid]
mgos_config_util.c:110  Extra key: [wrong_sta_protect.sta_pass]
mgos_config_util.c:110  Extra key: [wrong_sta_protect]
mgos_sys_config.c:368   MAC: e2:98:06:b5:7d:8d
mgos_sys_config.c:376   WDT: 30 seconds8��
mgos_deps_init.c:88     Init wifi 1.0...  r����8	�� #
esp_main.c:145          SDK: bcn 0                          �
esp_main.c:145          SDK: del if1                         r����8	�� #
esp_main.c:145          SDK: usl                                            �
esp_main.c:145          SDK: mode : null                                     r����8
mgos_wifi.c:88          WiFi STA: Using config 0 (JNET)                                 �� �
mgos_wifi.c:468         WiFi mode: STA                                                      �lrr�9l�Ğb��
esp_wifi.c:118          WiFi mode: STA                                                                  ��b��br�b����brc��
esp_main.c:145          SDK: mode : sta(e0:98:06:b5:7d:8d)                                                                ����l�x~�n�n����
esp_main.c:145          SDK: add if0                                                                                                      l���n��
esp_main.c:145          SDK: sleep disable                                                                                                       �
mgos_wifi.c:270         WiFi STA: Connecting to JNET                                                                                              �
mgos_deps_init.c:88     Init http-server 1.0...                                                                                                    l���
mgos_http_server.c:361  HTTP server started on [80]                                                                                                    �
mgos_deps_init.c:88     Init mbedtls 2.16.6-cesanta1...                                                                                                 �l`mgos_deps_init.c:88     Init ota-http-client 1.0...
mgos_deps_init.c:88     Init ota-http-server 1.0...
mgos_deps_init.c:88     Init rpc-common 1.0...
mgos_deps_init.c:88     Init rpc-service-config 1.0...
mgos_deps_init.c:88     Init rpc-uart 1.0...
mgos_rpc_channel_ua:313 0x3fff06bc UART0
mg_rpc.c:560            0x3fff06bc '' UART
mg_rpc.c:489            0x3fff06bc CHAN OPEN (UART UART0)
main.c:293              FW booted from app1, waiting for network connection
mgos_init.c:33          Init done, RAM: 52936 total, 46584 free, 46112 min free
mgos_event.c:134        ev MOS0 triggered 0 handlers
mgos_ota_core.c:991     1 0
mgos_ota_core.c:1329    UID: 035b81d28dc1badd, license: none
mgos_event.c:134        ev WFI1 triggered 0 handlers
mgos_net.c:86           WiFi STA: connecting
mgos_event.c:134        ev NET1 triggered 2 handlers
esp_main.c:145          SDK: scandone
[... cut ...]
main.c:106              Got chunk size: 536 received: 2467
main.c:106              Got chunk size: 536 received: 3003
main.c:106              Got chunk size: 536 received: 3539
main.c:106              Got chunk size: 536 received: 4075
main.c:121              spi_flash_erase_sector: 8.
main.c:128              spi_flash_write: 32768, 0x3fff0f5c, 4096.
main.c:135              spi_flash_write done
main.c:106              Got chunk size: 536 received: 4611
main.c:106              Got chunk size: 536 received: 5147
```

jimmyw, 2021-05-02:
@yaourdt Mark, do you have any idea? Not sure how to add additional logging.

juliogonzalez, 2021-05-19:
@jimmyw exactly the same problem here. How did you manage to open the plug and access the serial connectors?

Can you share a some pictures? I'd like to avoid breaking the case, but I could not find any examples.

juliogonzalez, 2021-05-19:
False alarm! I was alarmed by the fact that no LED was working... but eventually I realized that the Wifi for tasmota was present, and it seems it's working fine.

[image]

I am no expert on tasmota, but still if you want share how to open the plug, I can spare one to have a look and see if I can flash using a serial cable using esptool.py

juliogonzalez, 2021-08-13:
I tried to flash another two, one worked, the other one failed. I wasn't really checking if I could see the filesystem via HTTP.

Right now:
- Doesn't create its own Wifi network
- Doesn't connect to the old Wifi network
- The blue led is on for less than a second right after connecting the plug
- The red led is on but with very low intensity, so it's even hard to see

I will be happy to try a few things, if someone tells me how to open the plug without destroying it :-)

@jimmyw how did you manage?

juliogonzalez, 2021-08-13:
I feel stupid. It's just a matter of removing a phillips screw from the hole at the bottom of the smarplug.

After that, it's just a matter of following https://faulty.cloud/blog/shelly-plug-s-pinout (attaching the image with the pinout in case the blog is removed later):

[image]

If did not try to flash `sonoff.bin`, since I build a customized tasmota (it worked just fine):

```
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 ${HOME}/firmware.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e0:98:06:ab:8f:c4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 553920 bytes to 396771...
Wrote 553920 bytes (396771 compressed) at 0x00000000 in 35.5 seconds (effective 124.8 kbit/s)...
Hash of data verified.
```

And ready:
[image]

I have no clue of why this unit failed during the conversion. I'd not rule a human error (meaning I used the wrong URL for the firmware).

But in any case, if someone else is having the same issue I described at my previous comment (not sure if it's the same  @jimmyw had), fixing it is trivial as long as you have a FTDI adapter that provides 3,3v. No soldering is required if you are careful :-)

Ramalama2, 2023-03-04:
@juliogonzalez You're amazing, thank you!

It took 5 minutes to recover my briked shelly plug s.
I had the same issue as jimmi in the first post.

Cheers