yaourdt/mgos-to-tasmota #41: May brick the device

gshilo, 2021-05-03:
Hello

I tried this on Three Shelly 1L's.

The first worked perfectly, just as described. Restarted a couple of times and came back with Tasmota in just few minutes and was working great since.

Few days later I tried the exact same procedure on another Shelly 1L that was working fine with stock firmware. This time the device died. It gave me the json answer that says status=updating which is the normal behavior. Then it restarted and never came back. It was inaccessible  via wifi and the physical switch did not work either. I thought I did something wrong, so I tried with another one, after all the first one went so well.

Pointed it to the suitable url, this time double checking it's for Shelly 1L, checking that the link is alive and not touching anything during the process. Unfortunately  this one also never came back from its restart.

So I do not know exactly what went wrong in the Two last times and what made a difference from the first one, but I will have to flush them manually.

daroczig, 2022-01-27:
I had the same problem with Shelly 1: everything went fine with the first device, but the second failed (did not connect to WiFi after the automatic restart, neither seems to be working in AP mode as no Tasmota similar SSID showed up). I tried to manually restart a few times, also tried to make it connect to `mgos-recover`, but without luck.

h0tw1r3, 2022-02-02:
Very first device bricked.

TheNoim, 2022-09-12:
It also failed for a Shelly 1 and a Shelly 1L for me. Really annoying, because now I need to take it out of the wall again.