yaourdt/mgos-to-tasmota #44: Mgos2Tasmota Shelly Dimmer2 nothing happens

Fettkeewl, 2021-05-20:
Hello!

I think there needs to be some adjustment to the downloaded files
as nothing is happening on my shelly dimmer 2 when I try to OTA..
The link itself works because I can manually download the .zip-file.

This is the response i get
```
{"status":"updating","has_update":false,"new_version":"","old_version":"20200629-130515/release-1.7@28744b8b"}
```

It's missing "new version" as seen here
https://youtu.be/_oRr8FZyyQ0?t=393

flixlix, 2021-05-27:
Same issue on my Shelly 1 :(

Ramblurr, 2021-05-30:
Having this issue with a new out of box Shelly 2.5

`{"status":"updating","has_update":false,"new_version":"","old_version":"20191217-074901/1047-long-id-for-shelly-devices@ab2bdce7"}`

flixlix, 2021-05-31:
I solved my problem! My shelly was being blocked by the firwell, so it had no access to Internet. If you want to know if your shelly has access to the internet, you can move your phone out of the network (i.e. Mobile Data) and try toggling the Shelly, if it doesn't work, you might want to check your firewall blocking rules.

ManlioD, 2024-10-27:
If anyone hav this problem in 2024, i found this other githab repo that get the zip trati from tasmota:

https://github.com/alexdelprete/mgos-to-tasmota