yaourdt/mgos-to-tasmota #45: Bricked Shelly 1

jeroenst, 2021-05-23:
I tried to flash tasmota to a shelly 1 with the python script.

When I refreshed the webpage during flash it showed me a folder with a couple of files

Then it became unreachable and doesn't connect to the wifi anymore.

I also set up a mgos-recover ssid but it's not connecting to it and I also don't see an ssid of the shelly coming online.

```

stock_release_info:{'isok': True, 'data': {'SHPLG-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-1.zip', 'version': '20210115-103101/v1.9.4@e2732e05'}, 'SHPLG-S': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-S.zip', 'version': '20210429-101611/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG-S.zip', 'beta_ver': '20210514-070847/v1.10.5-rc1-gecc99b7'}, 'SHPLG-U1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-U1.zip', 'version': '20210429-101656/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG-U1.zip', 'beta_ver': '20210514-070929/v1.10.5-rc1-gecc99b7'}, 'SHPLG2-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG2-1.zip', 'version': '20210429-101742/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG2-1.zip', 'beta_ver': '20210514-071319/v1.10.5-rc1-gecc99b7'}, 'SHSW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-1.zip', 'version': '20210429-100340/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-1.zip', 'beta_ver': '20210514-065742/v1.10.5-rc1-gecc99b7'}, 'SHSW-21': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-21.zip', 'version': '20210429-100513/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-21.zip', 'beta_ver': '20210514-065912/v1.10.5-rc1-gecc99b7'}, 'SHSW-25': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-25.zip', 'version': '20210429-100559/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-25.zip', 'beta_ver': '20210514-065956/v1.10.5-rc1-gecc99b7'}, 'SHSW-PM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-PM.zip', 'version': '20210429-101828/v1.10.4-g3f94cd7'}, 'SHSW-L': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-L.zip', 'version': '20210429-100426/v1.10.4-g3f94cd7'}, 'SHAIR-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHAIR-1.zip', 'version': '20210429-101914/v1.10.4-g3f94cd7'}, 'SHSW-44': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-44.zip', 'version': '20201019-110845/v1.6.6_4Pro@bad181e4'}, 'SHUNI-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHUNI-1.zip', 'version': '20210429-102528/v1.10.4-g3f94cd7'}, 'SHEM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM.zip', 'version': '20210429-102232/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHEM.zip', 'beta_ver': '20210514-071759/v1.10.5-rc1-gecc99b7'}, 'SHEM-3': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM-3.zip', 'version': '20210429-102318/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHEM-3.zip', 'beta_ver': '20210514-071842/v1.10.5-rc1-gecc99b7'}, 'SHSEN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSEN-1_build.zip', 'version': '20200717-112934/v1.7.7@a02db0b8'}, 'SHMOS-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHMOS-01_build.gbl', 'version': '20210226-072307/v1.1.0@f31e1d2b'}, 'SHGS-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHGS-1.zip', 'version': '20210429-102444/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHGS-1.zip', 'beta_ver': '20210514-072008/v1.10.5-rc1-gecc99b7'}, 'SHSM-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSM-01.zip', 'version': '20210413-153317/v1.10.2-gb89901a'}, 'SHHT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHHT-1.zip', 'version': '20210413-154417/v1.10.2-gb89901a'}, 'SHWT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHWT-1.zip', 'version': '20210413-155706/v1.10.2-gb89901a'}, 'SHDW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-1.zip', 'version': '20210413-154021/v1.10.2-gb89901a'}, 'SHDW-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-2.zip', 'version': '20210413-154114/v1.10.2-gb89901a'}, 'SHSPOT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSPOT-1.zip', 'version': '20201124-091104/v1.9.0@57ac4ad8'}, 'SHCL-255': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHCL-255.zip', 'version': '20210429-095824/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHCL-255.zip', 'beta_ver': '20210514-065226/v1.10.5-rc1-gecc99b7'}, 'SHBLB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBLB-1.zip', 'version': '20210429-095954/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHBLB-1.zip', 'beta_ver': '20210514-065355/v1.10.5-rc1-gecc99b7'}, 'SHCB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHCB-1.zip', 'version': '20210429-095910/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHCB-1.zip', 'beta_ver': '20210514-065311/v1.10.5-rc1-gecc99b7'}, 'SHRGBW2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBW2.zip', 'version': '20210429-101353/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHRGBW2.zip', 'beta_ver': '20210514-070636/v1.10.5-rc1-gecc99b7'}, 'SHRGBWW-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBWW-01.zip', 'version': '20210429-100209/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHRGBWW-01.zip', 'beta_ver': '20210514-065611/v1.10.5-rc1-gecc99b7'}, 'SH2LED-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SH2LED-1.zip', 'version': '20210429-100254/v1.10.4-g3f94cd7'}, 'SHDM-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-1.zip', 'version': '20210429-102007/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHDM-1.zip', 'beta_ver': '20210514-071538/v1.10.5-rc1-gecc99b7'}, 'SHDM-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-2.zip', 'version': '20210429-102101/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHDM-2.zip', 'beta_ver': '20210514-071630/v1.10.5-rc1-gecc99b7'}, 'SHDIMW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDIMW-1.zip', 'version': '20210429-102146/v1.10.4-g3f94cd7'}, 'SHVIN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHVIN-1.zip', 'version': '20210429-100039/v1.10.4-g3f94cd7'}, 'SHBDUO-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBDUO-1.zip', 'version': '20210429-100125/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHBDUO-1.zip', 'beta_ver': '20210514-065526/v1.10.5-rc1-gecc99b7'}, 'SHBTN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBTN-1.zip', 'version': '20210415-130655/v1.10.3-g23074d0'}, 'SHBTN-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBTN-2.zip', 'version': '20210415-130749/v1.10.3-g23074d0'}, 'SHIX3-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHIX3-1.zip', 'version': '20210429-102402/v1.10.4-g3f94cd7', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHIX3-1.zip', 'beta_ver': '20210514-071924/v1.10.5-rc1-gecc99b7'}, 'SHPLG-UK1': {'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG-UK1.zip', 'beta_ver': '20210514-071128/v1.10.5-rc1-gecc99b7'}}}

intermediate_release_info:[['x2.9.1-.*', {'version': '2.9.1', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/issues/617', 'rel_notesX': '', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly1.zip', 'Shelly1L': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly1L.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly1PM.zip', 'Shelly2': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly2.zip', 'Shelly25': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly25.zip', 'ShellyI3': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyI3.zip', 'ShellyPlug': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyPlug.zip', 'ShellyPlugS': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyPlugS.zip', 'ShellyRGBW2': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyRGBW2.zip'}}], ['.*', {'version': '2.9.1', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.9.1', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly1.zip', 'Shelly1L': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly1L.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly1PM.zip', 'Shelly2': 'https:// [... cut ...] : 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-Shelly25.zip', 'ShellyI3': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyI3.zip', 'Shelly-i3': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyI3.zip', 'ShellyPlug': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyPlug.zip', 'ShellyPlugS': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyPlugS.zip', 'ShellyRGBW2': 'https://rojer.me/files/shelly/2.9.1/shelly-homekit-ShellyRGBW2.zip'}}]]

device_scan
```

deepcoder, 2021-05-25:
The one flaky plug US I have, I was able to get it back to showing the shelly ssid by doing mulitple power cycles. There is a power button on the plug US as well and during some of the power cycles I held the button down during power application, others times I pressed and held the button for 10 to 15 seconds after power application. I apologize that I do not have a more exact set of steps to getting back to the shelly setup ssid, but it did work on a unit that for all intent looked totally 'bricked'.

jeroenst, 2021-05-25:
I tried switching on and off multiple times without success, so I had to do a serial flash which succeeded.

closed 2021-05-25 by jeroenst