yaourdt/mgos-to-tasmota #49: Unable to Flash ShellyRGBW2

FlaMike, 2021-07-30:
I am trying to flash Tasmota on a stock Shelly RGBW2 with a 12 volt power supply.  The Shelly is on a network with internet access and I am using the correct script from your Github for the device.  I am able to download the .zip file to my PC that is on the same network as the Shelly.  I checked the "old version" of the Shelly software on my device and it matches the string below which I get when entering script URL.

{"status":"updating","has_update":false,"new_version":"","old_version":"20190822-083406/master@4148d2b7"}

The LED on the Shelly flashes a few times so it appears to be flashing and then flashes periodically, but the tasmota device never appears on my network and the Shelly does not update.

Any suggestions on how to do an OTA flash for this device are appreciated greatly.

Thank you!

oedm, 2021-08-24:
Facing the same issue. Also with 12 volt power supply and RGBW2 with current firmware from Shelly. HTTP Body is responding with:
`{"status":"updating","has_update":false,"new_version":"20210429-101525/v1.10.4-g3f94cd7","old_version":"20210429-101525/v1.10.4-g3f94cd7","beta_version":"20210820-121620/v1.11.4-rc2-gcc41d4b"}`

Sadly the Shelly just spikes shortly in ICMP response time but does not restart/flash the firmware.

Any advice to debug this behavior is appriciated.

Thank you so much!

oedm, 2021-08-24:
The debug log function pointed me to the right direction. I had used my RGBW2 already in white mode. I was able to OTA upgrade the firmware after switching back to RGB mode. :tada: 

Debug Log:
```
26401161 mgos_http_server.c:180  0x3fff2a3c HTTP connection from 192.168.178.47:60416
26441589 shelly_update.c:163     Starting ota from http://dl.dasker.eu/firmware/mg2tasmota-ShellyRGBW2.zip
26986748 mgos_ota_http_clien:270 Update URL: http://dl.dasker.eu/firmware/mg2tasmota-ShellyRGBW2.zip
27137353 mgos_ota_core.c:253     Starting, timeout 240, commit timeout 0, mem 32392
27207991 mgos_ota_core.c:487     FW: rgbw2-color esp8266 0.4.6 20210118-130933
27242566 shelly_update.c:90      BEGIN: rgbw2-color
27275228 shelly_update.c:98      will not update rgbw2-white with rgbw2-color
27309602 cloud_aes.c:169         Cloud already closed or not connected
27350652 mgos_ota_core.c:818     Update finished, result -13 (Aborted by event handler)
27391933 shelly_update.c:111     SHOTA: error Aborted by event handler
27426202 shelly_update.c:125     OTA is fubar
27467092 mgos_system.c:58        Rebooting in 500 ms
27513322 mgos_http_server.c:180  0x3fff2f2c HTTP connection from 192.168.178.47:60418
27563452 json.c:420              RAM: 49616 total, 32216 free
27949832 esp_main.c:137          SDK: state: 5 -> 0 (0)
27991406 esp_main.c:137          SDK: rm 0
28024336 esp_main.c:137          SDK: pm close 7
28062549 mgos_mqtt_conn.c:256    MQTT0 Disconnect
28095577 mgos_mqtt_conn.c:550    MQTT0 connecting after 1885 ms
28133922 mgos_wifi.c:119         WiFi STA: Disconnected, reason: 8
28176225 mgos_mdns.c:55          mDNS socket closed
28210026 mgos_mdns.c:82          Listening on udp://:5353
```

FlaMike, 2021-08-24:
So glad to see you were able to overcome the problem.  I've had failures with new out of the box units.  I will try switching them to white only & then back to RGB & see if that works.

Naive question...  How to you enable the debug log and where is it located?  I'm hoping it can help me just as it did you.  

Thanks for your help!

oedm, 2021-08-24:
I don't have any shelly with stock firmware currently at hand to verify, but I think you should find that at Settings -> Device Info.
Afterwards you can view the current log at `http://<SHELLY_IP>/debug/log`. Good luck!

FlaMike, 2021-08-24:
Thanks for your help!  I will try it later today.

Hakuren92, 2021-09-30:
I am unable to flash the RGBW2 with the `20210909-145215/v1.11.4-DNSfix-ge6b2f6d` firmware.
The debug log indicates a mismatch `will not update rgbw2-color with rgbw2`, and if I change to the white mode it will still mismatch between rgbw2-white and rgbw2.

Would it be enough to patch the name to make it match or could there be a risk that the device gets bricked somehow?

PaoloTK, 2023-04-10:
> I am unable to flash the RGBW2 running the `20210909-145215/v1.11.4-DNSfix-ge6b2f6d` firmware. The debug log indicates a mismatch `will not update rgbw2-color with rgbw2`, and if I change to the white mode it will still mismatch between rgbw2-white and rgbw2.
> 
> Would it be enough to patch the name to make it match or could there be a risk that the device gets bricked somehow?

Same issue here, will try modify the file and report back

scottboom, 2023-08-11:
@PaoloTK - Thanks for your comment. I was able to simply pull apart the mg2tasmota-ShellyRGBW2.zip file and update the manifest.json before carefully re-zipping. I say carefully, as the manifest has to be the first file in the .zip, and the .zip is not compressed. To re-create the zip on my BSD box, I used the command "zip -Z store -r mg2tasmota-ShellyRGBW2c.zip ./rgbw2-0.3.1/manifest.json ./rgbw2-0.3.1/esp_init_data_default_v08.bin ./rgbw2-0.3.1/fs.bin ./rgbw2-0.3.1/rboot.bin ./rgbw2-0.3.1/rgbw2.bin"