yaourdt/mgos-to-tasmota #60: Shelly Plus1

micahvandeusen, 2021-11-16:
It would be awesome if support could be added for the new Shelly Plus 1. The tasmota info is located here https://templates.blakadder.com/shelly_plus_1.html

peterforeman, 2021-11-19:
My Shelly1+ has this as info currently:

```
Device ID: A803xxxxxxxx
Firmware version: 0.8.1
Firmware build ID: 20210921-202758/0.8.1-g52de872
Web build ID: 1.5.4-8bcd3c1
```

I'm not sure which firmware file belongs to this.

fjh40, 2021-12-24:
Hi everyone, just flashed my Shelly Plus 1 devices via serial TTY to USB 3.3 volts a few days ago. Even though Shelly doesn't provide any information there's some information online about the pinout and which pins to connect to flash. Just make sure you use very small pins, I think 1mm2. Because normal jumper cables are way too big to fit in the Shelly Plus 1 for flashing purposes.

fl0wde, 2022-01-02:
> Hi everyone, just flashed my Shelly Plus 1 devices via serial TTY to USB 3.3 volts a few days ago. Even though Shelly doesn't provide any information there's some information online about the pinout [...]

Hey, can you explain and provide the pinout how you booted into flash mode of the Shelly Plus 1? 

Based on this page i was not able to understand how to be able to flash tasmota.

Thanks and in addition a mgos-to-tasmota for the new Shelly Plus' would be awesome :)

digiblur, 2022-01-02:
@fl0wde scan through here. I did the pin out and bit a reverse engineering and mistakes. 

https://youtu.be/eLoOT3mXcMs

fl0wde, 2022-01-02:
> @fl0wde scan through here. I did the pin out and bit a reverse engineering and mistakes.
> 
> https://youtu.be/eLoOT3mXcMs

Thanks, I saw the video already and at 8:42 you see the pinout in detail but i was not able to boot into flash mode even not with same procedure like the Shelly 1 by using GND at boot.

Would be great if you can share your experience and write 2-3 sentence to explain it to dumbs like me ;)

digiblur, 2022-01-02:
GPIO 0 needs to need held to ground as well. Nothing really special about it being esp32. Same flashing procedure as the esp8266.

fl0wde, 2022-01-02:
> GPIO 0 needs to need held to ground as well. Nothing really special about it being esp32. Same flashing procedure as the esp8266.

Thought so too, but obviously it's not working... what I'm doing wring here? :)

Here is how I set it up:

Shelly Plus 1	USB UART
Port 1		GND
Port 2		GND (at boot)
Port 3		-
Port 4		3V3
Port 5		TXD
Port 6		RXD
Port 7		-

fjh40, 2022-01-02:
@fl0wde,

Sorry for the vague photo, only one I can find since all my Shelly's are already built in. 

Shelly Plus 1	USB UART
Yellow		GND
Orange		GND 
Unused		-
Green		3V3
White		TXD
Red 		RXD
Unused		-

Above pin connections work on my devices, verify your USB UART device is 3.3 volts, instead of 5 volts.

[image]

fl0wde, 2022-01-05:
Thanks fjh40 and digiblur. Maybe I have just a issue by flasing the tasmota and pinout is fine...

How you are flashing the tasmota to the Shelly Plus 1 -  which tool?

I tried to flash with ESPEasy (ESPEasy_ESP32_mega-20211224) and failed all the time. When i tried with Espressif Flash Download Tool (Espressif_flash_download_tool_v3.8.5) I was able to flash the tasmota.

In addition I'm using the bin "tasmota32solo1.bin" version 10.1.0 (latest version), correct?

Sorry, but i'm really struggling with myself to identify the issue on my end :) Thanks guys!

micahvandeusen, 2022-01-05:
I had success with using these headers to make a custom cable for flashing https://www.amazon.com/dp/B00R1J4I6K?ref=ppx_pop_mob_ap_share. Then use the web based tasmota flasher

fl0wde, 2022-01-05:
> I had success with using these headers to make a custom cable for flashing https://www.amazon.com/dp/B00R1J4I6K?ref=ppx_pop_mob_ap_share. Then use the web based tasmota flasher and tasmota32 solo1.

Which web based tasmota flash you mean? I guess there is no web bases OTA image available for the Shelly Plus 1 and that's why this request was raised.

micahvandeusen, 2022-01-05:
This one https://tasmota.github.io/install/

MARGERM, 2022-02-12:
I now have installed the 3rd Shelly Plus1, three Shelly 1L and a Shelly Plug S.

Shelly 1L and the Shelly Plug S works fine with tasmota, using this update script for flashing them OTA.
I havn't any flashing tools, so it would be so helpful if you add the Shelly Plus1 to the list.

Current firmware:

Firmware version: 0.9.3
Firmware build ID: 20220117-162816/0.9.3-g354c141

MARGERM, 2022-02-12:
Just read, that the firmware 0.10.0-beta2 added this important thing:

**Support for /ota HTTP endpoint, for backward compatibility with Gen1 devices.**

oxc, 2022-02-12:
> Just read, that the firmware 0.10.0-beta2 added this important thing:
> 
> **Support for /ota HTTP endpoint, for backward compatibility with Gen1 devices.**

This is interesting, but I'm afraid this won't help too much. I believe the problem is that the mgos-to-tasmota currently does not even compile for ESP32 based devices. I needs to use a different library to write the downloaded firmware to the device's memory, so adding support for the Shelly Plus 1 (or any other ESP32-based device) requires a bit more work than adding another download URL.