yaourdt/mgos-to-tasmota #61: Adding Shelly Plus 1PM

hallard, 2021-11-20:
Just got my Shelly Plus 1P and I got same issue then #60 unable to find the matching firmware into the JSON file you provided
```
Firmware version: 0.8.1
Firmware build ID: 20210921-202918/0.8.1-g52de872
Web build ID: 1.5.4-8bcd3c1
```

So I enabled debug and fired a beta udate (0.9.2 beta-2) and here are the URL

```json
{
  "stable": {
    "version": "0.8.1",
    "build_id": "20210921-202918/0.8.1-g52de872",
    "url": "https://updates.shelly.cloud/fw/Plus1PM/9d370f43555de8353c761701e0e76bb58cf49f97233549168d9d8522579c2e1d"
  },
  "beta": {
    "version": "0.9.0-beta2",
    "build_id": "20211019-143747/0.9.0-beta2-gc6f3536",
    "url": "https://updates.shelly.cloud/fw/Plus1PM/51302a2460968dcf61ee290768930d8155da62172f3897a82008b6a7eb80efc7"
  }
}
```
If you need more info just don't hesitate to ask

segator, 2021-12-08:
Same issue here

tomm1ed, 2022-01-18:
Any update on this? I realize that it’s an ESP32 making it a different beast but would be nice to have. Smaller footprint and runs cooler than the ESP8266 version. Would love to be able to just OTA them

bobmel, 2022-01-27:
Looks to  me like the Shelly gen 2 API lacks upgrade API that allows url to be specified:
[https://shelly-api-docs.shelly.cloud/gen2/Overview/CommonServices/Shelly#shellyupdate](https://shelly-api-docs.shelly.cloud/gen2/Overview/CommonServices/Shelly#shellyupdate)

That will make ota upgrade to 3rd party firmware harder. Hope Shelly will add /ota api like gen 1 has.

MARGERM, 2022-02-12:
> Looks to me like the Shelly gen 2 API lacks upgrade API that allows url to be specified: https://shelly-api-docs.shelly.cloud/gen2/Overview/CommonServices/Shelly#shellyupdate
> 
> That will make ota upgrade to 3rd party firmware harder. Hope Shelly will add /ota api like gen 1 has.

Just added with firmware 0.10.0-beta2!
https://shelly-api-docs.shelly.cloud/gen2/changelog/

bobmel, 2022-02-21:
> Just added with firmware 0.10.0-beta2! https://shelly-api-docs.shelly.cloud/gen2/changelog/

Oh, that's really good.

If anyone has started to add support for Shelly Plus 1pm please let us know here. I'm considering taking a stab at it myself.

ohhai-kthxbai, 2022-03-05:
> 

Forgive my ignorance, but is upgrading to tasmota/esphome not just as simple as pointing that ota URL to a generic ESP32 .bin file? Is a new bootloader needed?

ThisIsTheOnlyUsernameAvailable, 2022-04-07:
> Has anyone already tried to flash it with the Shelly 1PM binary? Or is that not going to work at all? *I don't need bluetooth :-)

I believe the ESP8266 whereas the Plus 1 is an ESP32.

FerFemNemBem, 2022-05-02:
As the current Firmware of the Shelly plus 1PM has the OTA-API added and there seems to be a working Tasmota for this device ( https://templates.blakadder.com/shelly_plus_1PM.html => "tasmota32solo1.bin" ) - is there any progress on this issue?

ThisIsTheOnlyUsernameAvailable, 2022-05-02:
> As the current Firmware of the Shelly plus 1PM has the OTA-API added and there seems to be a working Tasmota for this device ( https://templates.blakadder.com/shelly_plus_1PM.html => "tasmota32solo1.bin" ) - is there any progress on this issue?

This firmware requires a direct serial connection to flash.  It works fine (I've done it with Tasmota & ESPHome), but this is very separate from the M2T process.

hagenbuch, 2022-06-06:
May I bump this issue up? I would also be very much interested in OTA Tasmota flashing, all the more for the Shelly plus 4 PM - which seems to have no (visible) pinheader for fashing. As we now seem to know which firmware works and the upload API seems to be there, hopefully it is not such a big issue.

Vorta, 2022-07-03:
Is it possible to update Shelly Plus 1PM with Tasmota or ESPHome firmware by just giving it the correct bin file in the OTA update or are there other things to look out for?

I'm following this issue because I have several Plus 1PMs which I want to migrate to ESPHome. I am very dissatisfied with Shelly firmware.

tomm1ed, 2022-08-15:
> I'm just here to confirm that the standard over the air (OTA) upgrading works flawlessly with a new 1 PM.
> 
> Remember that after you made the device upload the new software via the link and after some minutes of waiting, the Wifi configuration will be forgotten and you'll have to configure a "naked" Tasmota as we are used, using IP 192.168.4.1
> 
> So I guess this could be closed then?
> 
> Andreas

Are you sure you have the plus version? If so what link did you use since this GitHub still doesn’t mention any of the plus version. Just the regular ESP8266 based versions…

RobertoCravallo, 2022-09-25:
Any news? Flashed a 2.5 a couple of hours ago and that worked really nice. But what is the news on the 1 PM PLUS?
Regards from Germany

JakeShirley, 2022-10-28:
Popping my head in here to said I'd love this too!

sammyke007, 2022-11-09:
Any update on this?

JakeShirley, 2022-11-09:
I gave up waiting for this and just manually flashed my five new ones last night.

The only hard part is the 1.27mm pin headers.

[image]

mvgijssel, 2022-11-09:
Nice! I don’t have enough patience for the small pin headers so I bought this thing https://www.tindie.com/products/voltlog/voltlink-cp2102n-usb-serial-adapter-programmer/ which works really well!

JakeShirley, 2022-11-09:
@mvgijssel Does that adapter allow you to chose to bridge GPIO0 to GND or not? I like being able to power on the device in flash mode and normal mode.

mvgijssel, 2022-11-09:
> @mvgijssel Does that adapter allow you to chose to bridge GPIO0 to GND or not? I like being able to power on the device in flash mode and normal mode.

I think so, I think it does everything automatically. If you connect the adapter to the Shelly it boots and works as normal. And if you use the adapter to flash it will also work without any manual intervention.

sammyke007, 2023-01-09:
> I think so, I think it does everything automatically. If you connect the adapter to the Shelly it boots and works as normal. And if you use the adapter to flash it will also work without any manual [...]

Does the Voltlink Shelly adapter fit into the Shelly 1PM Plus please?

mvgijssel, 2023-01-10:
> Does the Voltlink Shelly adapter fit into the Shelly 1PM Plus please?

Yes it does! It provides an optional Shelly programming adapter. So it looks like this when the Voltlink is connected to the Shelly using the programming adapter:

[image]

albertomontesg, 2023-01-22:
I believe that support for the Shelly plus 1PM to flash OTA can be found here: https://github.com/tasmota/mgos32-to-tasmota32