yaourdt/mgos-to-tasmota #65: Update does not appear to start

Hoosier-IT, 2021-12-13:
http://192.168.1.223/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-Shelly1PM.zip

Gives: {"status":"updating","has_update":false,"new_version":"","old_version":"20200416-113902/???"}

Sympatron, 2022-03-20:
Same for me. 

Running `http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-Shelly25.zip`
gives me `{"status":"updating","has_update":false,"new_version":"","old_version":"20211109-125214/v1.11.7-g682a0db"}` for my Shelly 2.5.

rvanbaalen, 2022-06-24:
I have the same issue. I enter the correct URL but nothing happens.

Trevo525, 2022-11-11:
I also am having the same issue. The thing that I noticed is the "new_version":"" is empty. The tutorial linked by digiblur has something in that field. I wonder if this is the cause.

ohare93, 2022-11-13:
Same issue happened to me. It turned out the wifi settings I had on the Shelly meant it could not connect to the internet, and thus it could not download the new OS file. This was evident by the fact that it could not show the time at the top right, as it could not connect to time.google.com as is set in the settings. **Disabling a static ip fixed this for me** :+1:

Trevo525, 2022-11-13:
Confirmed on my end too. I just connected mine to my main wifi and did the update successfully..