yaourdt/mgos-to-tasmota #66: Shelly Plug S broken

pilaGit, 2021-12-14:
When Shelly Plug S link is input, it seems no new file exists on that address to be downloaded. Row with "new" remains empty in the page.

I did Bulb Duo and Shelly 1 with no problem.

pilaGit, 2021-12-15:
Funny thing. If I try just to download the file, it is OK. Just when I try OTA, no go. I had to resort to a screwdriver and TTL adapter.

grassroot, 2021-12-19:
Hello,
I can confirm the behaviour. The given OTA can be downloaded, but the upgrade does not start.
Here's my output: 
`{"status":"updating","has_update":false,"new_version":"","old_version":"20190516-073020/master@ea1b23db"}`
However, I was able to come up with a workaround. The "old" source at [dl.dasker.eu](http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip) seems to work!
I downloaded both, unzipped and took a look at insides of the zips:

http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyPlugS.zip :
```
$ unzip -l mg2tasmota-ShellyPlugS.zip
Archive:  mg2tasmota-ShellyPlugS.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
     1459  2020-12-04 13:20   shelly-plug-u1-0.3.1/manifest.json
      128  2020-12-04 13:20   shelly-plug-u1-0.3.1/esp_init_data_default_v08.bin
   262144  2020-12-04 13:20   shelly-plug-u1-0.3.1/fs.bin
     2320  2020-12-04 13:20   shelly-plug-u1-0.3.1/rboot.bin
   591280  2020-12-04 13:20   shelly-plug-u1-0.3.1/shelly-plug-u1.bin
---------                     -------
   857331                     5 files
```

http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip :
```
$ unzip -l mg2tasmota-ShellyPlugS.zip.1
Archive:  mg2tasmota-ShellyPlugS.zip.1
  Length      Date    Time    Name
---------  ---------- -----   ----
     1458  2021-01-18 13:08   shelly-plug-s-0.4.6/manifest.json
      128  2021-01-18 13:08   shelly-plug-s-0.4.6/esp_init_data_default_v08.bin
   262144  2021-01-18 13:08   shelly-plug-s-0.4.6/fs.bin
     2320  2021-01-18 13:08   shelly-plug-s-0.4.6/rboot.bin
   621120  2021-01-18 13:08   shelly-plug-s-0.4.6/shelly-plug-s.bin
---------                     -------
   887170                     5 files
```

Note the difference in folder naming. ota.tasmota.com version seems to pointo to "u1" (="US"?) version, while that dasker.eu to a "s"-version. According to [another source](https://github.com/mongoose-os-apps/shelly-homekit/issues/749), there are variations which may have differences between GPIO-usage.

So, while not understanding this all, I was able to flash Tasmota to my four Plug S without further issues using this dasker.eu URL like this: `http://<plug-IP-address>/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip`.
Hopefully this helps someone with similar issues :-)

matteofranceschini, 2021-12-24:
This makes perfectly sense, thanks for the warning.

Although, I'm here because sadly with my Shelly Plug S I get the same json.

```
{"status":"updating",
"has_update":false,
"new_version":"",
"old_version":"20190516-073020/master@ea1b23db"}
```

and it doesn't work with the dasker (http://<plug-IP-address>/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip) link :(

MARGERM, 2022-02-13:
Just worked February 3rd 2022 with my Shelly Plug S

volkerwestphal, 2022-04-23:
Still broken today.

JohnEridium, 2022-05-25:
Just ran into the same issue with both download sources:
`{"status":"updating","has_update":false,"new_version":"","old_version":"20190516-073020/master@ea1b23db"}` when I open `http://192.168.2.162/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip`

Newer source is the same:
`{"status":"updating","has_update":false,"new_version":"","old_version":"20190516-073020/master@ea1b23db"}`
opened: `http://192.168.2.162/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyPlugS.zip`

hagenbuch, 2022-06-06:
Everyone reading this: After applying the link with the .zip file to update to Tasmota, even when it works, the device will again lose its previously configured Wifi and start from scratch by offering an accesspoint - I overlooked this for quite some time!

Chose the Wifi ("Tasmota-something-number"), it does not need a password. Then browse to 192.168.4.1 ... that's all.

BiasF, 2022-10-07:
This destroyed my Shelly Plug S.
I got this message: 
{"status":"updating","has_update":false,"new_version":"","old_version":"20220209-094058/v1.11.8-g8c7bb8d"}

Since I used this URL, my Shelly Plug S isn't reachable via WIFI and also the button on it is without any function.
Turn it off and on again doesn't solve that. I have to open it for flashing via FTDI or throw it away.

hagenbuch, 2022-10-10:
I had this too but got my module running again. There seems to be a procedure of plugging and unplugging the device 6 times in half a minute to reset the entire configuration. From then on, the device should offer its  unprotected Wifi. Tasmota docs explains this special reset, good luck!

daloki83, 2022-10-19:
Same for me. Flashed 12/13 devices successfully today. One failed with the same behavior as described above. Also the 6times plug/unplug didn't help. Will try to flash it tomorrow with usb2ttl adapter.

pgorod, 2022-11-25:
For those having trouble getting the download to work, even from `dl.dasker.eu`, I am not sure of what I made to get it working, but I suspect that it was updating to the latest Shelly firmware before doing the OTA to Tasmota. So if you're desperate, try it (and tell us if it made a difference). 

Any way, my ShellyPlugS is fully Tasmoted now! 💪 🎉