yaourdt/mgos-to-tasmota #68: Shelly EM Not providing Monitoring Data Just Relay

ebarretopr, 2022-01-03:
Good Afternoon,

I have flashed a Shelly EM OTA, once it opened, I only have the On/Off control no power monitoring.

I have used the following templates with the option 21 enable on both without any luck, any help is appreciated.

{"NAME":"Shelly EM","GPIO":[0,0,288,0,32,0,0,0,640,3456,608,224,0,1],"FLAG":0,"BASE":18}

{"NAME":"Shelly EM","GPIO":[0,0,0,0,0,0,0,0,6,156,5,21,0],"FLAG":15,"BASE":18}

Thanks!

tobby88, 2022-04-12:
Disconnect power and reconnect. This should help.

hagenbuch, 2022-06-06:
Yeah, I had the same issue with a Shelly 2.5 - turning off power and then after 10 seconds reconnecting sorted things out quite "automagically".