yaourdt/mgos-to-tasmota #72: Shelly Plus i4

richcrabtree, 2022-03-30:
I would like to be able to flash the i4. I'm not certain what mongoose firmware it uses. How can I find out?

eric1905, 2022-04-23:
I am interested as well in the support for the plus i4

The output of /shelly:

{
name: null,
id: "shellyplusi4-a8032ab1f8a8",
mac: "A8032AB1F8A8",
model: "SNSN-0024X",
gen: 2,
fw_id: "20220330-151414/0.10.1-g8d603a2",
ver: "0.10.1",
app: "PlusI4",
auth_en: false,
auth_domain: null
}

I am not sure which firmware this is from https://api.shelly.cloud/files/firmware

According to blackadder a specific tasmota32 bin is needed

https://templates.blakadder.com/shelly_plus_i4.html

jmonsted, 2023-03-12:
I'm trying to flash the plus i4 DC. It appears to be the same as the i4, just with a different power supply. 

Output of /shelly:
{"name":null,"id":"shellyplusi4-xx","mac":"xx","model":"SNSN-0024X","gen":2,"fw_id":"20230308-091236/0.14.1-g22a4cb7","ver":"0.14.1","app":"PlusI4","auth_en":false,"auth_domain":null}

jmonsted, 2023-04-08:
Just went back to check and the i4 is now on the list. The upgrade worked flawlessly. Thanks!

kibeki, 2024-09-07:
anybody found out which flashfile to use for a Shelly Plus i4?