yaourdt/mgos-to-tasmota #73: Problem with conversion to has

TriderG75, 2022-04-03:
Hi, using this command http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2haa-Shelly1.zip, instead appearing the hotspot haa-xxxxxx it opens a mongoose page and the shelly 1 stops to work. Using the mg2tasmota command, all is ok. I add the photo of the mongoose page I'm talking about. 
Thanks

[image]

getinmybelly, 2022-04-14:
Same problem here, though with the Dimmer 2 update. 

[image]

jamr200, 2022-04-15:
I have the same problem with shelly1 and shelly 2.5

Tibi-rvb, 2022-04-19:
Same for me with a RGBW2

marcoscv-work, 2022-04-25:
Same thing here!
do you have any news about this?

RavenSystem, 2022-05-11:
@yaourdt GitHub did some changes in their SSL certificates and your firmware are not able to download files from its servers. Probably, a new build, compiled with latest SDK, is needed.

michapi, 2022-05-11:
Is there a way to recover the shelly?

mallito1307, 2022-05-11:
Same issue here!
Please solve it 🙏🏻

tokamac, 2022-05-23:
Wow, your program "installing" (failing to) HAA actually bricked the two Shelly Dimmer 2 modules I bought. Now they are impossible to reach or update. And you won't even provide an update, yet alone make a warning statement about HAA on the main install page. Shame. Refund! Ahah juste joking but you seriously pissed me off.

TriderG75, 2022-05-24:
> statement

You can use the method two messages above to recover the Shellys. Of course you have to use the correct file for your Shelly

tokamac, 2022-05-24:
Thanks, that worked! Using SHDM-2.zip for the Dimmer 2. I see that RavenSystem advises the following in the Third-Party Installation Tools on the HAA install page, for Shelly:

> [Shelly Devices OTA Version](https://github.com/yaourdt/mgos-to-tasmota)
IMPORTANT: Until developer fixes [current issue](https://github.com/yaourdt/mgos-to-tasmota/issues/73#issuecomment-1124240911), install Tasmota, then [Tasmota Minimal](http://ota.tasmota.com/tasmota/tasmota-minimal.bin.gz) from WebUI, and then [fullhaaboot.bin](https://github.com/RavenSystem/haa/releases/latest/download/fullhaaboot.bin) from WebUI.

I put the solution here since it's been weeks this annoying bug exists but the developper won't step in to make at least a statement.

abishur, 2022-06-03:
dumb question here, but when you ran the http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2haa-Shelly1.zip command did you replace the "shellyip" with the ip of your shelly?

TriderG75, 2022-06-04:
> dumb question here, but when you ran the http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2haa-Shelly1.zip command did you replace the "shellyip" with the ip of your shelly?

Of course. If you don't put the ip the command doesn't work and the Shelly remains with the original fw. In this case something changes in the fw and "bricks" the Shelly. Fortunately there is a way to recover it (look some messages above).

arkn666l, 2024-01-08:
for me works today, yes i saw this page .. index of.. but minutes later the tasmota wifi appears...