yaourdt/mgos-to-tasmota #74: flashing ESP32 based shelly devices

Kugeleis, 2022-04-05:
I see no ESP32 based device in the list of supported devices. Could e.g. [tasmota32solo1.bin](http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin) be added as used for Shelly Plus 1?

EDSTOBI, 2022-05-15:
Hallo
Ich habe versucht diese Datei selber zu erstellen wie im Mongoose OS quick start quide beschrieben, jedoch funktioniert diese nicht, da nach dem übertragen der shelly nicht mehr startet.
Auch ich würde mich über eine sauber kompetierte tasmota32solo1.zip freuen

hagenbuch, 2022-06-06:
Please let me also bump this up, I am also interested..

ssancese, 2022-11-17:
Hello, interested as well...

Kugeleis, 2022-11-17:
I flashed my shelly via serial. It's possible without opening and soldering.

sammyke007, 2023-01-09:
> I flashed my shelly via serial. It's possible without opening and soldering.

What pins did you use as it's very small?

Kugeleis, 2023-01-09:
> > I flashed my shelly via serial. It's possible without opening and soldering.
> 
> What pins did you use as it's very small?

See https://templates.blakadder.com/shelly_plus_1.html for a picture with named PINs. There you have RX/TX for serial access. Yes, the PIN holes are tiny. I used thin wire.

Washee, 2023-02-19:
> I flashed my shelly via serial. It's possible without opening and soldering.

This may work for many Shelly devices, but not for all.
If I am not mistaken devices like the Shelly plug products dont offer an accessible serial pinheader.