yaourdt/mgos-to-tasmota #76: Support for Shelly 3EM

philrich, 2022-04-24:
Hi,

As of Version 11.1.0 of Tasmota Shelly 3EM is fully supported (Firmware has to be compiled manually), so I decided to convert the last one of my Shelly Devices to Tasmota. All my other Shelly Devices could be flashed with Tasmota OTA via mgos-to-tasmota (Thanks for that!) unfortunately Shelly 3EM is missing as supported Device. 

I managed to build the firmware myself with attached patch applied and it worked.
[mgos-to-tasmota-shelly3em.patch.gz](https://github.com/yaourdt/mgos-to-tasmota/files/8550673/mgos-to-tasmota-shelly3em.patch.gz)

After installing the default Tasmota firmware a custom version has to be built as the driver for the Shelly 3EM is not included in the pre-built firmware, at least not in Version 11.1.0

Important! Don't forget to download the calibration file of the original Shelly Firmware before flashing Tasmota! (`curl http://<shelly-ip>/calib.dat -o calib.dat`). See https://templates.blakadder.com/shelly_3EM.html for instructions.

Step by step guide:
  * Download calib.dat
  * Build mgos-to-tasmota with attached patch applied:
    `mos build --build-var MODEL=Shelly3EM --build-var TARGETFW=tasmota --platform esp8266`
  * Start tiny webserver in `build/fw` directory (`python3 -m http.server 8000`) or copy to real webserver which is accessible by Shelly
  * Start flashing process. Enter `http://<shellyip>/ota?url=http://<webserver-ip>/shellyem3-esp8266-0.4.6.zip`
  * Attach Shelly to WIFI as usual
  * Build Tasmota with Shelly 3EM driver included. I used this guide: https://github.com/arendst/Tasmota/discussions/13515#discussioncomment-2536926
  * Done

hagenbuch, 2022-08-15:
Great, thanks! I'd strongly support this being integrated into the project, also the PM4 Pro!

slimcdk, 2022-12-10:
Any chance that you have found a solution to extract this using the UART? The Shelly 3EM runs Mongoose OS, but it seems that they switch the debug UART from UART0 to UART1 after boot. Usually you can do RPC over UART with MGOS, but not sure if they have enabled it and if there is a way to access UART1.

zeminator, 2022-12-30:
Also would like to see fully ready OTA firmware for this.

hagenbuch, 2022-12-31:
@zeminator I don't know what you mean but @philrich describes exactly that in just the post above you, he even offers a first binary! From that binary, you may flash whichever other Tasmota via browser but it will have to include the (compiled in) code for the measurement chip built into the 3EM

I used the Tasmota online compiler to make my version including that sensor, just be bold and trust that all these messages don't mean a lot and after a minute you've compiled your build..

linuxdep, 2023-01-16:
Hi, is the output of the API exactly the same as the original? Some projects rely on the direct query from the 3EM. With this version, is the correct calculation made for balancing meters?

hagenbuch, 2023-01-17:
Not sure what you mean by API. It is known that the original firmware allows us to "download" file calib.dat from the device. They contain calibration data for the three phases, phase angles etc. This data must be slightly cleaned up as I understand it and then converted to a rule, the rules makes Tasmota take in the calibration variables. I did that but I never was sure if the calibration data had an effect.

linuxdep, 2023-01-17:
> 

By API I mean calling 3EM-IP/status

11notes, 2023-11-08:
> Hi,
> 
> As of Version 11.1.0 of Tasmota Shelly 3EM is fully supported (Firmware has to be compiled manually), so I decided to convert the last one of my Shelly Devices to Tasmota. All my other Shelly Devices could be flashed with Tasmota OTA via mgos-to-tasmota (Thanks for that!) unfortunately Shelly 3EM is missing as supported Device.
> 
> I managed to build the firmware myself with attached patch applied and it worked. [mgos-to-tasmota-shelly3em.patch.gz](https://github.com/yaourdt/mgos-to-tasmota/files/8550673/mgos-to-tasmota-shelly3em.patch.gz)
> 
> After installing the default Tasmota firmware a custom version has to be built as the driver for the Shelly 3EM is not included in the pre-built firmware, at least not in Version 11.1.0
> 
> Important! Don't forget to download the calibration file of the original Shelly Firmware before flashing Tasmota! (`curl http://<shelly-ip>/calib.dat -o calib.dat`). See https://templates.blakadder.com/shelly_3EM.html for instructions.
> 
> Step by step guide:
> 
>     * Download calib.dat
> 
>     * Build mgos-to-tasmota with attached patch applied:
>       `mos build --build-var MODEL=Shelly3EM --build-var TARGETFW=tasmota --platform esp8266`
> 
>     * Start tiny webserver in `build/fw` directory (`python3 -m http.server 8000`) or copy to real webserver which is accessible by Shelly
> 
>     * Start flashing process. Enter `http://<shellyip>/ota?url=http://<webserver-ip>/shellyem3-esp8266-0.4.6.zip`
> 
>     * Attach Shelly to WIFI (Connect to tasmota-xxx SSID, then go to http://192.168.4.1)
> 
>     * Build Tasmota with Shelly 3EM driver included. I used this guide: [Support for Shelly 3EM arendst/Tasmota#13515 (reply in thread)](https://github.com/arendst/Tasmota/discussions/13515#discussioncomment-2536926)
> 
>     * Install Minimal Tasmota, then upgrade to custom built Firmware
> 
>     * Voila

Thank you for everything, thanks to you I could create this container image that will compile the firmware: https://hub.docker.com/r/11notes/mgos-to-tasmota

11notes, 2023-11-08:
> Also important: Many forget it, I forgot it: Even after a correct update. the sensors will not show up if you don't switch power off and on.

Ah no problem this is pretty clear.

jeroenst, 2024-11-30:
I can confirm this patch worked perfectly for me and I will make a pull request.

Here is the prebuild firmware for shelly 3em: [shellyem3-esp8266-0.4.6.zip](https://github.com/user-attachments/files/17965055/shellyem3-esp8266-0.4.6.zip)

plodocus, 2025-01-01:
Did you flash while the Shelly was connected to mains (AC)?

jeroenst, 2025-01-01:
Yes

core66, 2025-03-06:
> Yes, ota

only with the help of the update function on the original Shelly interface?`I just have to upload your zip and the Shelly is then flashed OTA with Tasmota?

Sorry, but I have to ask very carefully, because I won't be able to get to the Shelly 3EM so easily.

jeroenst, 2025-03-06:
I believe I flashed the 3em with the shelly 1 firmware from mgos to tasmota and flashed the rebuild shelly 3em firmware afterwards via the tasmota firmware update button.

11notes, 2025-03-07:
> > Yes, ota
> 
> only with the help of the update function on the original Shelly interface?`I just have to upload your zip and the Shelly is then flashed OTA with Tasmota?
> 
> Sorry, but I have to ask very carefully, because I won't be able to get to the Shelly 3EM so easily.

@core66 https://hub.docker.com/r/11notes/mgos-to-tasmota