yaourdt/mgos-to-tasmota #80: Shelly Plug responds "Not Found" when trying to flash

oester, 2022-07-22:
I picked up a couple of Shelly Plugs, and was trying to flash them using  the same process I did before:

http://192.168.30.133/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugUS.zip

But the device responds "Not Found" in the browser.

Device ID: 083AF2006F44
Firmware version: plugusprod2
Firmware build ID: 20220211-132652/plugusprod2_app-gcb4621f
Web build ID: 1.5.5-c01b44e

This worked fine on the plugs I picked up early last year. Perhaps the firmware is an issue.

BinaryBurgr, 2022-07-30:
Same thing happens with mine; found out it's because Shelly is now shipping the second generation model called the Shelly **_Plus_** Plug US (notice the `prod2` in the firmware version). The shell is the same shape (now a matte finish though), but the internals were completely redesigned. Hardware wise [it's a Shelly Plus 1PM in plug form](https://www.reddit.com/r/ShellyUSA/comments/veuto4/comment/ict1j05/?utm_source=share&utm_medium=web2x&context=3).

This means that it needs a different mgos-to-tasmota firmware package, which I'm fairly certain doesn't exist yet as it doesn't appear to support any of the second generation "**Plus**" devices. I _imagine_ the OTA URL has moved to http://deviceip/#/ota like the rest of the webui, but I didn't want try it with a link to any incompatible firmware for fear of bricking it. I don't mind sticking with the Shelly firmware on my new ones for now, they're apparently more stable than the old Plug US (not that I had issues with those on the Shelly firmware either).

fredmaria, 2022-11-06:
Can you please elaborate on the url  http://deviceip/#/ota.
What is the meaning of #
Thanks in advance.