yaourdt/mgos-to-tasmota #83: Shelly 1PM "old" model vers. "new" Model

RobertoCravallo, 2022-09-25:
First of, way cool project, simply brilliant!!!!

Does the firmware for the "old" 1PM work with the "new" model? They do look like two completely different devices, so I am not sure!

DNicholai, 2022-10-26:
I have the same question.  It appears as though the Shelly 1PM plus is ESP32 based versus ESP8266.  Therefore unlikely to work as a drop in replacement.

RobertoCravallo, 2022-10-26:
> I have the same question. It appears as though the Shelly 1PM plus is ESP32 based versus ESP8266. Therefore unlikely to work as a drop in replacement.

Hi, I found out (don't remember where) that you have to flash them as SOLO1!!! Works with the web based install, no problems. Just scroll down the list until you hit SOLO1!!!

sanderlv, 2022-10-30:
> Hi, I found out (don't remember where) that you have to flash them as SOLO1!!! Works with the web based install, no problems. Just scroll down the list until you hit SOLO1!!!

Where and what list?

sammyke007, 2023-01-09:
Did it work for the Shelly 1PM plus?

RobertoCravallo, 2023-01-09:
> Did it work for the Shelly 1PM plus?

Yes it did. With this [https://tasmota.github.io/install/](url)