yaourdt/mgos-to-tasmota #85: Wrong firmware zipped inside ShellyPlugS.zip?

pgorod, 2022-11-24:
Hi. Thanks for your work, it is soooo useful. 🎉 

I hope I'm reporting this in the right place 🤞 

I'm trying to flash Tasmota onto a Shelly Plug S device using this URL:

http://mydevice.ip.address/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyPlugS.zip

But it won't work. The log shows:

```
90424714 shelly_update.c:168     Starting ota from http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyPlugS.zip
90931061 mgos_ota_http_clien:276 Update URL: http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyPlugS.zip
91192674 mgos_ota_core.c:268     Starting, timeout 240, commit timeout 0, mem 37532
91213773 mgos_ota_core.c:652     FW: shelly-plug-u1 esp8266 0.3.1 20201204-132056 sig 0x00
91220315 shelly_update.c:90      BEGIN: shelly-plug-u1
91225518 shelly_update.c:98      will not update shelly-plug-s with shelly-plug-u1
91230761 cloud_aes.c:162         Closing 0x3fff1bac
91236153 mgos_ota_core.c:1050    Update finished, result -13 (Aborted by event handler)
91252658 shelly_update.c:111     SHOTA: error Aborted by event handler
91257652 shelly_update.c:130     OTA is fubar
```

And if I download the zip at `http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyPlugS.zip` onto my PC and open it, I see it has a single directory called `shelly-plug-u1-0.3.1` in it. Shouldn't that be `shelly-plug-s` or something like that? Is it possible that the wrong zip got uploaded by mistake?

Thanks for any help!

pgorod, 2022-11-25:
@arendst

Is this the correct place to open this issue? Or are the files in `ota.tasmota.com` coming from `https://github.com/arendst/mgos-to-tasmota/` instead? Any way, in your repo I can't open issues..

I noticed that people have faced this problem earlier: https://github.com/yaourdt/mgos-to-tasmota/issues/66  and as noted there, in fact the firmware works fine if I download it from `dl.dasker.eu` instead. Maybe this is as simple as copying the correct file into the `ota.tasmota.com` server? 

Thanks!

arendst, 2022-11-27:
@pgorod 

I just replaced the "broken" PlugS on ``ota.tasmota.com`` by copying it from ``dl.dasker.eu``.

The goal of the ``ota.tasmota.com`` repository was/is to provide shelly firmware replacements using the latest released tasmota binaries. The mgos zips were adjusted for that except the PlugS one. The dl.dasker.eu did, at that time not support the latest released tasmota binariues.

closed 2022-11-27 by pgorod