yaourdt/mgos-to-tasmota #86: Tasmota on Shelly Plug S not working?

luk6173, 2022-11-26:
Hi,

i used http://192.168.0.xxx/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyPlugS.zip to install Tasmota on my Shelly Plug S and waited around 5 Minutes but the Plug is not working now. After power cycle the plug blinks very short blue that's all. I don't see any Tasmota AP from my Android and iPhone. What I can do now to get my Plug running?

Thank you,
Lukas

muhlba91, 2022-12-15:
i just ran into the same issue...

gempir, 2022-12-19:
I converted 3 plugs successfully. 

My 4th one doesn't wanna work though. I saw the mongoose webserver in my browser shortly but now it doesn't connect to the old Wifi and it doesn't provide a tasmota-whatever Wifi either. 

I tried creating the recovery network but it didn't log in there either. I'm unsure if there is anything left I can do or if its just bricked now and I have to somehow connect to it wired? Which is probably not a great idea on a 230v device...