yaourdt/mgos-to-tasmota #93: Not able to use ota, update always fails, is the verison to old?

TKaluza, 2023-02-08:
Hello!
I tried differenty ways to flash a Shelly RGBW2 ota. I also hosted a small server for downloading the file via my network without https and a static ip but I always get:
"http://192.168.178.24/ota?url=http://192.168.178.100:8080/mg2tasmota-ShellyRGBW2.zip"
{"status":"updating","has_update":false,"new_version":"20221027-100516/v1.12.1-ga9117d3","old_version":"20221027-100516/v1.12.1-ga9117d3"}

and it seems to me it will not use the provided "older" firmware. But I do not know. 
Any suggestions what I'm doing wrong?

andy71, 2023-02-18:
Same here for the Shelly Duo RGBW through the original link.

`{"status":"updating","has_update":false,"new_version":"20221027-085547/v1.12.1-ga9117d3","old_version":"20221027-085547/v1.12.1-ga9117d3"}`

TheFes, 2023-05-26:
Same here for a Shelly Dimmer 2

Julusian, 2023-06-11:
@TKaluza I just tried this on a RGBW2 and found that it was rejecting the firmware because the name didn't match. It was trying to write `rgbw2` firmware when it needed to be called `rgbw2-color`.
This RGBW2 isn't new, perhaps in a recent shelly firmware they changed this?

I recompiled the RGBW2 firmware with that name changed and it flashed successfully for me:
[shelly-rgbw2.zip](https://github.com/yaourdt/mgos-to-tasmota/files/11715550/shelly-rgbw2.zip)

PR for the fix https://github.com/arendst/mgos-to-tasmota/pull/2