yaourdt/mgos-to-tasmota #95: http://dl.dasker.eu is unreachable

SebastianoDiCecio, 2023-02-16:
http://dl.dasker.eu is unreachable

Tobi266, 2023-02-17:
Has someone downloaded the zip-files before and can share it?

flo269, 2023-02-17:
> Has someone downloaded the zip-files before and can share it?

https://github.com/yaourdt/mgos-to-tasmota/releases/tag/release-v0.4.6

Tymose, 2023-02-18:
You can also change command if you are using int from: 
"- http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyXX.zip"
"+ http://shellyip/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyXX.zip"

See fork https://github.com/alexdelprete/mgos-to-tasmota for proper command (it is old but working).

mkroeber, 2023-03-11:
Same here... i tryed http://192.168.178.90/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-Shelly1PM.zip with the result: "Not found"

The tasmota link seems to work.. at least it downloads the zip-file...
The Shelly link seems to work, it opens the dashboard

What am i doing wrong?

LordMike, 2023-03-12:
I came here because I'm trying to flash my Shelly Duo RGBW, of which I've already bricked two, but I wanted to try again.

I also found that Dasker was down, but luckily, all firmwares are available on a release, thanks for the link. I had troubles starting the flashing though, because specifically for ShellDuoRGBW, tasmota.com doesn't have it. I tried linking directly to the Github.com zip file, but even in http, it redirects to https, which I think Mongoose has an issue with.

So I started up a webserver (I use [dotnet-serve](https://github.com/natemcmaster/dotnet-serve)) locally, and served the file through that.

1. Open a terminal in the directory with your zip file
2. Run `dotnet-serve -a 0.0.0.0 -p 80` to start up a HTTP server on port 80
3. Browse to it in a browser to verify it works, use your LAN IP, e.g. `http://192.168.0.100/`
4. Get the link to your zip file, and build up the ota url like such: `http://192.168.0.130/ota?url=http://192.168.0.100/mg2tasmota-ShellyMYDEVICE.zip`

Hope this helps others in the future who struggle to serve a zip file over http :)

tmonxx, 2023-03-26:
> You can also change browser command if you are using it. From: "- http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyXX.zip" To: "+ http://shellyip/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyXX.zip"
> 
> See fork https://github.com/alexdelprete/mgos-to-tasmota for proper command (it is old but working).

Is done! Thank You :)

owenhankey, 2023-03-29:
> I came here because I'm trying to flash my Shelly Duo RGBW, of which I've already bricked two, but I wanted to try again.
>
> I also found that Dasker was down, but luckily, all firmwares are available [...]

Hi, i tried using dontnet serve but it only gave me the ip of http://localhost. i tried changing the ip address but got errors

Unexpected error: System.Net.Sockets.SocketException (49): Can't assign requested address

how do i get a local ip address rather than local host?

LordMike, 2023-03-29:
You need to find your LAN IP address, this is typically of the form `192.168.x.x` or `10.x.x.x`. Use `ipconfig` to see what your address is.

owenhankey, 2023-03-30:
Thanks! i got it. it tried to flash but it didnt work and now its bricked. but thanks for your help. gonna try opening it up and flashing it directly

LordMike, 2023-04-14:
Enable debug logging on the shelly page (under settings somewhere).. Do the update, wait a little (20s or so), and then download the log from the same page. It should contain hints as to what's up.

thunder312, 2023-06-25:
Hi folks,
I just tried to flash my Shelly RGBW2 with Tasmota.
The links was (using tasmota, because dasker is offline):
http://192.168.178.75/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip
Is it possible, that the pre-built patch isn't suitable?
Because as far as I read the log (below), the Shelly accepts only patches with "rgbw2- color" or "rgbw2-white".
Yes the shelly can handle 1 color led stripe or 4 white led stripes.
I tried both modes, before patching.
Do I need to build my own patch? Please say no. :-/

**The JSON output was:** 
_{"status":"updating","has_update":false,"new_version":"20230511-223416/v1.13.1-gb0b8c19","old_version":"20230511-223416/v1.13.1-gb0b8c19"}_

**As you can see:**
old version = new version, which is the newest Shelly default firmware.

**The LOG:**
_122687570:1687699693973873 mgos_sys_config.c:323   Saved to conf9.json
122967750 mgos_http_server.c:180  0x3fff2a04 HTTP connection from 192.168.178.98:50679
125867680 mgos_http_server.c:180  0x3fff2a04 HTTP connection from 192.168.178.98:50680
125887472 json.c:426              RAM: 50936 total, 35220 free
130219455 mgos_http_server.c:180  0x3fff2b34 HTTP connection from 192.168.178.98:50681
130227358 shelly_update.c:168     Starting ota from http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip
130253967 mgos_http_server.c:180  0x3fff2b34 HTTP connection from 192.168.178.98:50682
130733276 mgos_ota_http_clien:276 Update URL: http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip
130895693 mgos_ota_core.c:268     Starting, timeout 240, commit timeout 0, mem 34416
130936458 mgos_ota_core.c:652     FW: rgbw2 esp8266 0.3.1 20201204-114410 sig 0x00
130940819 shelly_update.c:90      BEGIN: rgbw2
**130944970 shelly_update.c:98      will not update rgbw2-white with rgbw2**
130949113 cloud_aes.c:165         Cloud already closed or not connected
130954564 mgos_ota_core.c:1050    Update finished, result -13 (Aborted by event handler)
130965639 shelly_update.c:111     SHOTA: error Aborted by event handler
130969646 shelly_update.c:130     OTA is fubar
130974832 mgos_system.c:58        Rebooting in 500 ms
131429191 esp_main.c:138          SDK: state: 5 -> 0 (0)
131434069 esp_main.c:138          SDK: rm 0
131438529 esp_main.c:138          SDK: pm close 7
131446738 mgos_wifi.c:70          WiFi STA: Disconnected, reason: 8
131452463 esp_main.c:138          SDK: sleep enable,type: 2
131458280 mgos_net.c:84           WiFi STA: disconnected
131463387 shelly.c:312            WiFi status=1313166336; mode=1
131469132 coiot.c:491             Coap listener 0x3fff2274 is closed!
131474397 coiot.c:616             coap_connect, WIFI status=1313166336 mode=1_