yaourdt/mgos-to-tasmota #100: http://dl.dasker.eu/ not reachable

Omega236, 2023-04-20:
The links are offline.

Has someone an alternative link?

bibixx, 2023-04-21:
@Omega236 I have found this repo to function correctly: https://github.com/tasmota/mgos32-to-tasmota32 (tested on Shelly Plus 2 PM)
If you can't see the `Custom Firmware` in the Shelly's options, update the Shelly's built in firmware to the latest version first

gcardy, 2023-04-22:
I wish that yaourdt had simply uploaded the binaries here to Github before stopping support of the project. :/

bibixx, 2023-04-22:
@gcardy I haven't tested it out but aren't all of the binaries here: [https://github.com/yaourdt/mgos-to-tasmota/releases/tag/release-v0.4.6](https://github.com/yaourdt/mgos-to-tasmota/releases/tag/release-v0.4.6)?

gcardy, 2023-04-23:
@bibixx Thank you for pointing that out.  I clearly did not look closely enough before posting.

mihadoo, 2023-04-30:
Then someone should update the links in https://github.com/yaourdt/mgos-to-tasmota readme.md from 

http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-Shelly1.zip

to

http://shellyip/ota?url=https://github.com/yaourdt/mgos-to-tasmota/releases/download/release-v0.4.6/mg2tasmota-Shelly1.zip

helmo, 2023-05-04:
There is a duplicatie issue in #95 

To summarize, the [original author stopped](https://github.com/yaourdt/mgos-to-tasmota/commit/9d945c5bd975b44a393660a363b2059fa2fe5f06) and others have forked. 

The one [from Tasmota's creator](https://github.com/arendst/mgos-to-tasmota) seems to be the logical choice.