yaourdt/mgos-to-tasmota #102: Hi folks,

thunder312, 2023-06-30:
Hi folks,
I just tried to flash my Shelly RGBW2 with Tasmota.
The links was (using tasmota, because dasker is offline):
http://192.168.178.75/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip
Is it possible, that the pre-built patch isn't suitable?
Because as far as I read the log (below), the Shelly accepts only patches with "rgbw2- color" or "rgbw2-white".
Yes the shelly can handle 1 color led stripe or 4 white led stripes.
I tried both modes, before patching.
Do I need to build my own patch? Please say no. :-/

**The JSON output was:** 
_{"status":"updating","has_update":false,"new_version":"20230511-223416/v1.13.1-gb0b8c19","old_version":"20230511-223416/v1.13.1-gb0b8c19"}_

**As you can see:**
old version = new version, which is the newest Shelly default firmware.

**The LOG:**
_122687570:1687699693973873 mgos_sys_config.c:323   Saved to conf9.json
122967750 mgos_http_server.c:180  0x3fff2a04 HTTP connection from 192.168.178.98:50679
125867680 mgos_http_server.c:180  0x3fff2a04 HTTP connection from 192.168.178.98:50680
125887472 json.c:426              RAM: 50936 total, 35220 free
130219455 mgos_http_server.c:180  0x3fff2b34 HTTP connection from 192.168.178.98:50681
130227358 shelly_update.c:168     Starting ota from http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip
130253967 mgos_http_server.c:180  0x3fff2b34 HTTP connection from 192.168.178.98:50682
130733276 mgos_ota_http_clien:276 Update URL: http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip
130895693 mgos_ota_core.c:268     Starting, timeout 240, commit timeout 0, mem 34416
130936458 mgos_ota_core.c:652     FW: rgbw2 esp8266 0.3.1 20201204-114410 sig 0x00
130940819 shelly_update.c:90      BEGIN: rgbw2
**130944970 shelly_update.c:98      will not update rgbw2-white with rgbw2**
130949113 cloud_aes.c:165         Cloud already closed or not connected
130954564 mgos_ota_core.c:1050    Update finished, result -13 (Aborted by event handler)
130965639 shelly_update.c:111     SHOTA: error Aborted by event handler
130969646 shelly_update.c:130     OTA is fubar
130974832 mgos_system.c:58        Rebooting in 500 ms
131429191 esp_main.c:138          SDK: state: 5 -> 0 (0)
131434069 esp_main.c:138          SDK: rm 0
131438529 esp_main.c:138          SDK: pm close 7
131446738 mgos_wifi.c:70          WiFi STA: Disconnected, reason: 8
131452463 esp_main.c:138          SDK: sleep enable,type: 2
131458280 mgos_net.c:84           WiFi STA: disconnected
131463387 shelly.c:312            WiFi status=1313166336; mode=1
131469132 coiot.c:491             Coap listener 0x3fff2274 is closed!
131474397 coiot.c:616             coap_connect, WIFI status=1313166336 mode=1_

Thanks for helping,

Daniel

_Originally posted by @thunder312 in https://github.com/yaourdt/mgos-to-tasmota/issues/95#issuecomment-1606097356_

wgroenewald, 2023-07-31:
+1
Same issue, either get 
```will not update rgbw2-white with rgbw2```
or
```will not update rgbw2-color with rgbw2```

chrishick, 2023-08-03:
Same issue, same FW version 20230511-223325/v1.13.1-gb0b8c19

pedymaster, 2023-09-05:
Hello,
I repacked the archive and it should work now, if you are on the rgbw2-color mode. Link you can use is here:
http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip

ChristiaanSl, 2023-09-15:
> Hello, I repacked the archive and it should work now, if you are on the rgbw2-color mode. Link you can use is here: http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip

How did you repack? Can you also repack for the Dimmer2? Many thanks in advance.

pedymaster, 2023-09-17:
Depending on what is the problem with dimmer2

In this case, I had to unzip the version that was failing with `will not update rgbw2-color with rgbw2`. There is a `manifest.json` file, where I had to replace `rgbw2` with `rgbw2-color` to fool the original shelly.
This needs to be carefully repacked again. Carefully, because the files then have to be repacked and the `manifest.json` has to be the first file in new zip. On linux it goes like `zip -Z store -r mg2tasmota-ShellyRGBW2c.zip ./rgbw2-0.3.1/manifest.json ./rgbw2-0.3.1/esp_init_data_default_v08.bin ./rgbw2-0.3.1/fs.bin ./rgbw2-0.3.1/rboot.bin ./rgbw2-0.3.1/rgbw2.bin`

So if only the renaming is needed, here is your manual :-) I have found this on some german forum. Not much else I can do in case it does not work.

thunder312, 2023-09-28:
@pedymaster Good solution and very easy. 🥇

bbccdd, 2023-09-28:
That worked great indeed pedymaster.

In my case it had to be **rgbw2-white** instead of color that did the trick.

it-guru, 2023-12-10:
I have the same problem as bbccdd - i also need a image for **rgbw2-white** . I already have take a look at mos tools, to build a image by my own - but i wasn't able to get mos running on my debian.
Please help!

pedymaster, 2023-12-10:
And what exactly are you missing in this thread?
Either before flashing configure your shelly rgbw2 to color mode instead of white (and then you can use my image) or just replace `rgbw2-color` for `rgbw2-white` in my steps and make your own

it-guru, 2023-12-13:
Oh, ok. 
Thanks for the info - but i have already flash it with a FTDI Adapter. I will try it on the next device (i have some in the ToDo list :-).

ChrisVonFrosta, 2024-03-29:
@pedymaster there are 2places with `rgbw2`. which one need to change in `rgbw2-color`?
`"name": "rgbw2",`
`"src": "rgbw2.bin"`

pedymaster, 2024-04-03:
Hey,
I believe its just the name, you can double check it in the image I
provided - http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip (or you can
just use that one)

gianlucagd53, 2024-04-07:
HI everyone, 
im new here and I wish to choose the right forum.
I have a shelly RGBW2 and I try to install tasmota by ota

I use the link 
http://shellyip/ota?url=http://dl.dasker.eu/firmware/mg2tasmota-ShellyRGBW2.zip
but it doesn't work
here the message

{
    "status": "updating",
    "has_update": false,
    "new_version": "20230913-113259/v1.14.0-gcb84623",
    "old_version": "20221013-134509/v1.12-1-g1abbae93c-rgbw2-prod",
    "beta_version": "20231107-164040/v1.14.1-rc1-g0617c15"
}

Can someone help me??? 

Thank you everyone

leoguiders, 2024-04-10:
> HI everyone, im new here and I wish to choose the right forum. I have a shelly RGBW2 and I try to install tasmota by ota
>
> I use the link [...]

The dl.dasker.eu web host with the firmwares is down, try
http://shellyip/ota?url=http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip for rgbw2 in color mode

gianlucagd53, 2024-04-10:
> The dl.dasker.eu web host with the firmwares is down, try http://shellyip/ota?url=http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip for rgbw2 in color mode

THANK YOU SO MUCH!
That's perfect

Kinggrass, 2024-07-04:
Hi tigether,
I also want to flash my rgbw2 but nothing is happening ,,, :/
Tried both urls:
[http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip](http://192.168.178.75/ota?url=http://ota.tasmota.com/tasmota/shelly/mg2tasmota-ShellyRGBW2.zip)
and
[http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip](http://shellyip/ota?url=http://dl.nekohell.eu/mg2tasmota-ShellyRGBW2c.zip)
without success :/

My RGBW2 is in color mode.

Its my first shelly, how can I get the log?
Under Logs over the shelly web ui nothing appears.

fallingcats, 2025-05-01:
The "new" link doesn't work anymore because it redirects to https and the shelly can't handle it. You can use mine instead
`http://shelly_ip/ota?url=http://get.fallingcats.net/mg2tasmota-ShellyRGBW2c.zip`