yaourdt/mgos-to-tasmota #115: Usable with ESPHome ?

vincegre, 2025-01-24:
Hi

Fantastic tool you made but is it usable same way using ESPHome firmware ? I prepare the bin file with ESPHome, zip it and pass URL at Shelly. Any chance it works or not ?
Also should I use old or new flash format when I compile in ESPHome for it ?

Thanks

Vincèn

TrentHouliston, 2025-02-11:
I've successfully used it with esphome, it required some modifications though. You build this repository but change it so that it looks for an esphome bin.

I added something like this to mos.yml

```
  - when: build_vars.TARGETFW == "esphome"
    apply:
      config_schema:
        - [
            "mg2x.url",
            "s",
            "http://192.168.23.254/local/mgos2esphome/esphome.bin",
            { "title": "URL of target firmware" },
          ]
```

and put my build esphome binary file in `www/mgos2esphome/esphome.bin` (my HA servers IP is 192.168.23.254).

Then the firmware you built here (the fw.zip) is downloaded onto the shelly device, and that firmware once it boots looks for that URL to get the esphome firmware which it then installs.

vincegre, 2025-02-12:
Hi @TrentHouliston 

Thanks for your help but I discovered in meantime that this repository should be shut down at it's no more usable with Shelly devices since at least 2/3 years after some fw updates from Shelly :(

Vincèn

closed 2025-02-12 by vincegre